mwyborski/Linux-Magic-Trackpad-2-Driver #1: Installation instructions

anton-johansson, 2017-07-21:
I tried to get this working in my Kubuntu 17.04 dist with the following:
1. Cloned the Kernel from `git://kernel.ubuntu.com/ubuntu/ubuntu-zesty.git`
2. Applied your changes
3. Built it and installed it [this way](https://wiki.ubuntu.com/Kernel/BuildYourOwnKernel) and rebooted.

Is there anything else I need to do to make it work? I'm quite lost after this, it feels like I need to do something more. I still only get the single finger movement and taps from the trackpad.

robotrovsky, 2017-08-31:
@anton-johansson you should make sure, that the correct driver bcm5974 is loaded for the device. Maybe the instructions here will help you:
https://github.com/robbi5/magictrackpad2-dkms

closed 2017-08-31 by robotrovsky

chenliushan, 2018-06-07:
Appreciate your work. I have upgraded my Ubuntu to 18.04, and installed this driver through `dkms`. However, the I still only get the single finger movement and taps from the trackpad too. Also, I have tried to copy the `hid-magicmouse.ko` from update/ to kernel/driver/hid and reboot, but still nothing change. 
Did I miss anything? Or any suggestions?

robotrovsky, 2018-06-07:
@chenliushan please try to load the compiled module with modprobe, then we will look why it fails:

1. Go to the linux/drivers/hid folder and "make" the module
2. "rmmod hid_magicmouse" to unload the module, if the original hid-magicmouse module was loaded
3. modprobe hid-magicmouse.ko
4. With "dmesg" you should see if the driver gets loaded when you connect the Magic Trackpad 2

Please tell me if this works. If so you maybe only need to add a entry to /etc/modules-load.d/. Maybe you also need to xz the kernel module. I dont know for sure, as i dont run ubuntu. But the manual loading of the driver should work like described.

/etc/modules-load.d/hid-magicmouse.conf:
`hid-magicmouse`

chenliushan, 2018-06-07:
Thanks for your reply. 
With the compiled module I still only get the single finger feature. However, I found some errors and exceptions in the "dmesg" which is attached as follows (not sure whether it is related) :
```
[    3.155981] mei_me 0000:00:16.0: enabling device (0000 -> 0002)
[    3.191211] ACPI Error: Needed [Buffer/String/Package], found [Integer] 000000005cca8dc6 (20170831/exresop-593)
[    3.191217] ACPI Exception: AE_AML_OPERAND_TYPE, While resolving operands for [Index] (20170831/dswexec-461)
[    3.191225] 
               Initialized Local Variables for Method [WVPO]:
[    3.191225]   Local1: 00000000d492f86a <Obj>           Buffer(12) 00 00 00 00 00 00 00 00
[    3.191234] Initialized Arguments for Method [WVPO]:  (2 arguments defined for method invocation)
[    3.191235]   Arg0:   00000000c00ad114 <Obj>           Integer 0000000000000004
[    3.191239]   Arg1:   000000005cca8dc6 <Obj>           Integer 0000000000000000
[    3.191244] ACPI Error: Method parse/execution failed \_SB.WMIV.WVPO, AE_AML_OPERAND_TYPE (20170831/psparse-550)
[    3.191251] ACPI Error: Method parse/execution failed \_SB.WMIV.WMPV, AE_AML_OPERAND_TYPE (20170831/psparse-550)
[    3.191414] RAPL PMU: API unit is 2^-32 Joules, 5 fixed counters, 655360 ms ovfl timer
[    3.191415] RAPL PMU: hw unit of domain pp0-core 2^-14 Joules
[    3.191416] RAPL PMU: hw unit of domain package 2^-14 Joules
[    3.191416] RAPL PMU: hw unit of domain dram 2^-14 Joules
[    3.191417] RAPL PMU: hw unit of domain pp1-gpu 2^-14 Joules
[    3.191417] RAPL PMU: hw unit of domain psys 2^-14 Joules
[    3.193335] ACPI Error: Needed [Buffer/String/Package], found [Integer] 000000009562973e (20170831/exresop-593)
[    3.193340] ACPI Exception: AE_AML_OPERAND_TYPE, While resolving operands for [Index] (20170831/dswexec-461)
[    3.193346] 
               Initialized Local Variables for Method [WVPO]:
[    3.193347]   Local1: 00000000ed8a4f47 <Obj>           Buffer(12) 00 00 00 00 00 00 00 00
[    3.193354] Initialized Arguments for Method [WVPO]:  (2 arguments defined for method invocation)
[    3.193355]   Arg0:   00000000ad9b2c82 <Obj>           Integer 0000000000000004
[    3.193357]   Arg1:   000000009562973e <Obj>           Integer 0000000000000000
[    3.193361] ACPI Error: Method parse/execution failed \_SB.WMIV.WVPO, AE_AML_OPERAND_TYPE (20170831/psparse-550)
[    3.193367] ACPI Error: Method parse/execution failed \_SB.WMIV.WMPV, AE_AML_OPERAND_TYPE (20170831/psparse-550)
[    3.194489] ACPI Error: Needed [Buffer/String/Package], found [Integer] 00000000061ce9a4 (20170831/exresop-593)
[    3.194494] ACPI Exception: AE_AML_OPERAND_TYPE, While resolving operands for [Index] (20170831/dswexec-461)
[    3.194500] 
               Initialized Local Variables for Method [WVPO]:
[    3.194501]   Local1: 00000000c11316c6 <Obj>           Buffer(12) 00 00 00 00 00 00 00 00
[    3.194510] Initialized Arguments for Method [WVPO]:  (2 arguments defined for method invocation)
[    3.194510]   Arg0:   000000004c392fdf <Obj>           Integer 0000000000000004
[    3.194514]   Arg1:   00000000061ce9a4 <Obj>           Integer 0000000000000000
[    3.194520] ACPI Error: Method parse/execution failed \_SB.WMIV.WVPO, AE_AML_OPERAND_TYPE (20170831/psparse-550)
[    3.194527] ACPI Error: Method parse/execution failed \_SB.WMIV.WMPV, AE_AML_OPERAND_TYPE (20170831/psparse-550)
[    3.194570] input: HP WMI hotkeys as /devices/virtual/input/input12
[    3.197574] ACPI Error: Attempt to CreateField of length zero (20170831/dsopcode-168)
[    3.197580] 
               Initialized Local Variables for Method [WVPI]:
[    3.197581]   Local0: 000000009626e20a <Obj>           Integer 0000000000000080
[    3.197584]   Local1: 00000000e5ff90c3 <Obj>           Package 00000000e5ff90c3
[    3.197587] Initialized Arguments for Method [WVPI]:  (3 arguments defined for method invocation)
[    3.197587]   Arg0:   00000000b520b003 <Obj>           Integer 0000000000000000
[    3.197589]   Arg1:   000000008b307c61 <Obj>           Integer 0000000000000003
[    3.197592]   Arg2:   0000000094e40f65 <Obj>           Buffer(20) 53 45 43 55 01 00 00 00
[    3.197598] ACPI Error: Method parse/execution failed \_SB.WMIV.WVPI, AE_AML_OPERAND_VALUE (20170831/psparse-550)
[    3.197606] ACPI Error: Method parse/execution failed \_SB.WMIV.WMPV, AE_AML_OPERAND_VALUE (20170831/psparse-550)

```

robotrovsky, 2018-06-08:
@chenliushan please make sure that the correct hid-magicmouse driver gets loaded and try the following steps:

    cd linux/drivers/hid
    make
    sudo rmmod hid_magicmouse
    sudo insmod ./hid-magicmouse.ko
    tail -f ~/.local/share/xorg/Xorg.0.log
Now unplug the trackpad and plug it back in, to see which driver gets loaded.

chenliushan, 2018-06-13:
Thanks @robotrovsky. This is what I got after running commands you have mentioned above:
```
[   140.574] (II) event20 - Apple Inc. Magic Trackpad 2: device is a pointer
[   140.574] (II) event20 - Apple Inc. Magic Trackpad 2: device removed
[   140.574] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.1/0003:05AC:0265.0009/input/input24/event20"
[   140.574] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: MOUSE, id 18)
[   140.576] (**) Option "AccelerationScheme" "none"
[   140.578] (**) Apple Inc. Magic Trackpad 2: (accel) selected scheme none/0
[   140.578] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[   140.578] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[   140.579] (II) event20 - Apple Inc. Magic Trackpad 2: is tagged by udev as: Mouse
[   140.579] (II) event20 - Apple Inc. Magic Trackpad 2: device is a pointer
[   741.920] (II) config/udev: removing device Apple Inc. Magic Trackpad 2
[   741.920] (**) Option "fd" "58"
[   741.920] (II) event20 - Apple Inc. Magic Trackpad 2: device removed
[   741.928] (II) UnloadModule: "libinput"
[   741.928] (II) systemd-logind: releasing fd for 13:84
[   748.103] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/mouse2)
[   748.103] (II) No input driver specified, ignoring this device.
[   748.103] (II) This device may have been added with another device file.
[   748.213] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event20)
[   748.214] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "evdev pointer catchall"
[   748.214] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput pointer catchall"
[   748.214] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "Touchpads"
[   748.214] (II) LoadModule: "mtrack"
[   748.214] (WW) Warning, couldn't open module mtrack
[   748.214] (II) UnloadModule: "mtrack"
[   748.214] (II) Unloading mtrack
[   748.214] (EE) Failed to load module "mtrack" (module does not exist, 0)
[   748.214] (EE) No input driver matching `mtrack'
[   748.214] (II) Falling back to input driver `libinput'
[   748.214] (II) Using input driver 'libinput' for 'Apple Inc. Magic Trackpad 2'
[   748.264] (II) systemd-logind: got fd for /dev/input/event20 13:84 fd 68 paused 0
[   748.265] (**) Apple Inc. Magic Trackpad 2: always reports core events
[   748.265] (**) Option "Device" "/dev/input/event20"
[   748.265] (**) Option "_source" "server/udev"
[   748.266] (II) event20 - Apple Inc. Magic Trackpad 2: is tagged by udev as: Mouse
[   748.266] (II) event20 - Apple Inc. Magic Trackpad 2: device is a pointer
[   748.266] (II) event20 - Apple Inc. Magic Trackpad 2: device removed
[   748.267] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.1/0003:05AC:0265.000D/input/input25/event20"
[   748.267] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: MOUSE, id 18)
[   748.269] (**) Option "AccelerationScheme" "none"
[   748.271] (**) Apple Inc. Magic Trackpad 2: (accel) selected scheme none/0
[   748.271] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[   748.271] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[   748.272] (II) event20 - Apple Inc. Magic Trackpad 2: is tagged by udev as: Mouse
[   748.272] (II) event20 - Apple Inc. Magic Trackpad 2: device is a pointer
```

robotrovsky, 2018-06-19:
@chenliushan have you been able to resolve this issue? Which distribution are you using?

mmulvahill, 2018-06-22:
@robotrovsky I'm having a similar problem with no input driver being loaded.  Pointer and left-click work, but no two-finger gestures.  

I'm using Linux Mint 19 beta at the moment and experiencing similar results on pure Ubuntu 18.04.

After doing the rmmod/insmod routine, I sget this in my Xorg log:

```
[  1664.705] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/mouse2)
[  1664.705] (II) No input driver specified, ignoring this device.
[  1664.705] (II) This device may have been added with another device file.
[  1664.787] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event7)
[  1664.787] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput pointer catchall"
[  1664.787] (II) Using input driver 'libinput' for 'Apple Inc. Magic Trackpad 2'
[  1664.787] (**) Apple Inc. Magic Trackpad 2: always reports core events
[  1664.787] (**) Option "Device" "/dev/input/event7"
[  1664.787] (**) Option "_source" "server/udev"
[  1664.876] (II) event7  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Mouse
[  1664.876] (II) event7  - Apple Inc. Magic Trackpad 2: device is a pointer
[  1664.876] (II) event7  - Apple Inc. Magic Trackpad 2: device removed
[  1664.908] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.4/1-10.4:1.1/0003:05AC:0265.0027/input/input25/event7"
[  1664.908] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: MOUSE, id 13)
[  1664.909] (**) Option "AccelerationScheme" "none"
[  1664.910] (**) Apple Inc. Magic Trackpad 2: (accel) selected scheme none/0
[  1664.910] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[  1664.910] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[  1664.968] (II) event7  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Mouse
[  1664.968] (II) event7  - Apple Inc. Magic Trackpad 2: device is a pointer
```

MeanderingCode, 2018-06-25:
I am seeing the same thing on Ubuntu. No mtrack, using libinput. I have seen an error about failed verification sometimes, but not when I manually insmod.

```
Jun 25 10:21:17 elessar kernel: Bluetooth: HIDP (Human Interface Emulation) ver 1.2
Jun 25 10:21:17 elessar kernel: Bluetooth: HIDP socket layer initialized
Jun 25 10:21:17 elessar kernel: hid-generic 0005:004C:0265.0006: unknown main item tag 0x0
Jun 25 10:21:17 elessar kernel: input: Magic Trackpad 2 as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256/0005:004C:0265.0006/input/input20
Jun 25 10:21:17 elessar kernel: hid-generic 0005:004C:0265.0006: input,hidraw5: BLUETOOTH HID v0.70 Mouse [Magic Trackpad 2] on e0:9d:31:da:51:5a
Jun 25 10:21:17 elessar kernel: PKCS#7 signature not signed with a trusted key
Jun 25 10:21:17 elessar kernel: hid_magicmouse: loading out-of-tree module taints kernel.
Jun 25 10:21:17 elessar kernel: hid_magicmouse: module verification failed: signature and/or required key missing - tainting kernel
Jun 25 10:21:17 elessar /usr/lib/gdm3/gdm-x-session[1628]: (II) config/udev: Adding input device Magic Trackpad 2 (/dev/input/mouse2)
Jun 25 10:21:17 elessar /usr/lib/gdm3/gdm-x-session[1628]: (II) No input driver specified, ignoring this device.
```

chenliushan, 2018-07-09:
Sorry, @robotrovsky I have not solved this problem.  
I am using Ubuntu to 18.04.

chenliushan, 2018-08-30:
My problem solved by upgrading the Ubuntu kernel to 4.18.5-041805-generic. 
(Although I have updated the Ubuntu to 18.04 earlier, somehow it still uses the old version kernel 4.15.)
So after I upgrade my kernel, this driver works.