mwyborski/Linux-Magic-Trackpad-2-Driver #2: Two sides of the trackpad not responsive

peterychuang, 2017-08-29:
Hi @robotrovsky, thank you for the great work!

I've been using the Magic Trakpad 2 with cable on Linux (I patch the kernel with [the patch you submitted to linux-input](https://patchwork.kernel.org/patch/9831283/)) for a bit over an afternoon, and it mostly works fine. However, I notice that sizeable portions on both sides of the touchpad aren't responsive. In those areas, click is registered, but not tap. Tracking doesn't work if the finger input begins from either side of the trackpad, but it works if the movement begins from the middle of the trackpad and moves to the sides.

Let me know if there's anything I can do to help fix the problem. Thanks again for your work!

peterychuang, 2017-08-30:
It's probably related: the trackpad's size in mm reported in ```libinput list-devices``` is wrong:

```
Device:           bcm5974
Kernel:           /dev/input/event4
Group:            6
Seat:             seat0, default
Size:             104x76mm
Capabilities:     pointer 
Tap-to-click:     disabled
Tap-and-drag:     enabled
Tap drag lock:    disabled
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   *two-finger edge 
Click methods:    button-areas *clickfinger 
Disable-w-typing: enabled
Accel profiles:   none
Rotation:         n/a
```

And I've just noticed that part of the top of the trackpad isn't responsive in the way I described in the previous message.

peterychuang, 2017-08-30:
On further investigation, I find out that this is in fact a problem with ```libinput```, not your driver.

Thanks again for your work. I'm closing this issue.

closed 2017-08-30 by peterychuang

robotrovsky, 2017-08-31:
Hi @peterychuang !

Thank you. Good to know, that you sorted out the problems. Sadly i got no response at all about the official patch. Would be nice to see it in mainline, but doesn't seem like it will happen soon :-/

peterychuang, 2017-09-01:
@robotrovsky, Yes, it does take quite a bit of persistence to get patches merged into mainline.

Just curious, will you be working on getting the trackpad working via Bluetooth?

robotrovsky, 2017-09-01:
I think that should not be too hard, as the protocol is the same. But i won't spend more time on the driver, if i don't get any feedback.

ponyfleisch, 2017-09-10:
@peterychuang for bluetooth, have a look at my driver here: https://github.com/ponyfleisch/hid-magictrackpad2

peterychuang, 2017-09-10:
@ponyfleisch thanks for your work! though do you happen to have the version of the driver that I can use to patch the kernel directly instead of dkms?

ponyfleisch, 2017-09-10:
@peterychuang not at the moment. can you just run make in usr/src/hid-magictrackpad2-4.10+hid-magictrackpad2/ and do `rmmod hid-generic; insmod ./hid-magictrackpad2.ko` to test if it works generally?