mwyborski/Linux-Magic-Trackpad-2-Driver #5: quirks.c and 4.14.43

poisonnuke, 2018-06-05:
when trying to compile to the rasbian kernel, I receive the following errors:

`In file included from drivers/hid/hid-core.c:34:0:
drivers/hid/hid-core.c:1982:42: error: ‘USB_DEVICE_ID_ASUSTEK_T100_KEYBOARD’ undeclared here (not in a function)
  { HID_USB_DEVICE(USB_VENDOR_ID_ASUSTEK, USB_DEVICE_ID_ASUSTEK_T100_KEYBOARD) },
                                          ^
./include/linux/hid.h:651:47: note: in definition of macro ‘HID_USB_DEVICE’
  .bus = BUS_USB, .vendor = (ven), .product = (prod)
                                               ^~~~
drivers/hid/hid-core.c:2033:41: error: ‘USB_DEVICE_ID_ELECOM_DEFT_WIRED’ undeclared here (not in a function)
  { HID_USB_DEVICE(USB_VENDOR_ID_ELECOM, USB_DEVICE_ID_ELECOM_DEFT_WIRED) },
                                         ^
./include/linux/hid.h:651:47: note: in definition of macro ‘HID_USB_DEVICE’
  .bus = BUS_USB, .vendor = (ven), .product = (prod)
                                               ^~~~
drivers/hid/hid-core.c:2034:41: error: ‘USB_DEVICE_ID_ELECOM_DEFT_WIRELESS’ undeclared here (not in a function)
  { HID_USB_DEVICE(USB_VENDOR_ID_ELECOM, USB_DEVICE_ID_ELECOM_DEFT_WIRELESS) },
                                         ^
./include/linux/hid.h:651:47: note: in definition of macro ‘HID_USB_DEVICE’
  .bus = BUS_USB, .vendor = (ven), .product = (prod)
                                               ^~~~
drivers/hid/hid-core.c:2035:41: error: ‘USB_DEVICE_ID_ELECOM_HUGE_WIRED’ undeclared here (not in a function)
  { HID_USB_DEVICE(USB_VENDOR_ID_ELECOM, USB_DEVICE_ID_ELECOM_HUGE_WIRED) },
                                         ^
./include/linux/hid.h:651:47: note: in definition of macro ‘HID_USB_DEVICE’
  .bus = BUS_USB, .vendor = (ven), .product = (prod)
                                               ^~~~
drivers/hid/hid-core.c:2036:41: error: ‘USB_DEVICE_ID_ELECOM_HUGE_WIRELESS’ undeclared here (not in a function)
  { HID_USB_DEVICE(USB_VENDOR_ID_ELECOM, USB_DEVICE_ID_ELECOM_HUGE_WIRELESS) },
                                         ^
./include/linux/hid.h:651:47: note: in definition of macro ‘HID_USB_DEVICE’
  .bus = BUS_USB, .vendor = (ven), .product = (prod)
                                               ^~~~
scripts/Makefile.build:328: recipe for target 'drivers/hid/hid-core.o' failed
`

possible, that some device IDs are mixed up because of different kernel versions? What is the best way to fix this?

robotrovsky, 2018-06-05:
Hi @poisonnuke ,
this is due to changes in the kernel:
https://github.com/torvalds/linux/commit/a9ef00aec7633b3be3b19a7fce9a8d13f5a07c1c#diff-0fd4976957ef951b8b4cd61bc6d90199

To use the driver with the older kernel, you would have to make a diff to 4.17-rc7 for the three files in linux/drivers/hid and then patch your older kernel accordingly.

poisonnuke, 2018-06-05:
I see, 4.14.43 does not have a hid-quirks.c at all, so I could savely delete it?

robotrovsky, 2018-06-05:
The content was in hid-core.c as far as i remember. But you can ignore this file. Actually you check out the project and try the DKMS. It ignores hid-quirks.c as well.

Please give feedback, if it worked with the DKMS and on the RPI.

poisonnuke, 2018-06-05:
It is nearly impossible to backport the hid-magicmouse.c to 4.14. Tooo many errors in depending files like input.h, hid.h, input-event-codes.h and some more.

Is there any way, I could get the hid_hw_raw_request to work inside a usb_driver? Because I would like to stay with the bcm5974

robotrovsky, 2018-06-05:
Why dont you try then the modified bcm5974 ?

https://github.com/robotrovsky/linux/blob/master/drivers/input/mouse/bcm5974.c
https://github.com/robotrovsky/linux/blob/master/drivers/hid/hid-apple.c
https://github.com/robotrovsky/linux/blob/master/drivers/hid/hid-core.c
https://github.com/robotrovsky/linux/blob/master/drivers/hid/hid-ids.h

poisonnuke, 2018-06-05:
as my question indicates, the bcm5974 is a usb-driver, the hid-magictrackpad is a hid-driver and I need the hid_hw_raw_request from the hid-trackpad driver (for the haptic feedback) but cant find any way myself to get it work (hid_device = usb_get_intfdata(iface) produces a segfault))

any suggestions on this?

robotrovsky, 2018-06-05:
The Request ID's are different between USB and bluetooth. To do a hid_hw_raw_request look in the bcm5974 how usb_control_msg is used:

	data = kmalloc(c->um_size, GFP_KERNEL);
	if (!data) {
		dev_err(&dev->intf->dev, "out of memory\n");
		retval = -ENOMEM;
		goto out;
	}

	/* read configuration */
	size = usb_control_msg(dev->udev, usb_rcvctrlpipe(dev->udev, 0),
			BCM5974_WELLSPRING_MODE_READ_REQUEST_ID,
			USB_DIR_IN | USB_TYPE_CLASS | USB_RECIP_INTERFACE,
			c->um_req_val, c->um_req_idx, data, c->um_size, 5000);

	if (size != c->um_size) {
		dev_err(&dev->intf->dev, "could not read from device\n");
		retval = -EIO;
		goto out;
	}

	/* apply the mode switch */
	data[c->um_switch_idx] = on ? c->um_switch_on : c->um_switch_off;

	/* write configuration */
	size = usb_control_msg(dev->udev, usb_sndctrlpipe(dev->udev, 0),
			BCM5974_WELLSPRING_MODE_WRITE_REQUEST_ID,
			USB_DIR_OUT | USB_TYPE_CLASS | USB_RECIP_INTERFACE,
			c->um_req_val, c->um_req_idx, data, c->um_size, 5000);

	if (size != c->um_size) {
		dev_err(&dev->intf->dev, "could not write to device\n");
		retval = -EIO;
		goto out;
}

poisonnuke, 2018-06-06:
I guess I need some differnt IDs for that.
tried that:
`size = usb_control_msg(dev->udev, usb_sndctrlpipe(dev->udev, 0),
		BCM5974_WELLSPRING_MODE_WRITE_REQUEST_ID,
		USB_DIR_OUT | USB_TYPE_CLASS | USB_RECIP_INTERFACE,
		cfg->um_req_val, cfg->um_req_idx, enableHostClick, sizeof(enableHostClick), 5000);`

but no error but trackpad-click is still on. How could I find the appropriate IDs? Looks like I need a MacBook and some tool to intercept USB traffic?

robotrovsky, 2018-06-06:
[recorded_data.zip](https://github.com/robotrovsky/Linux-Magic-Trackpad-2-Driver/files/2075667/recorded_data.zip)
In this file you will find sniffed data for different actions. It is recorded using the MT2 with OSX using a Beagle 480. The data-center software to view the sniffs can be downloaded for free from the totalphase website. Maybe this helps you.

poisonnuke, 2018-06-06:
thank you this is indeed very good, but at this point Im sort of too n00b to understand what is happening. The requests are totally different for the same action and I have no clue what is the requestID and so on. I think Im out, I do not have the ressources at the moment to fully grasp the usb-protocol from scratch. sry

robotrovsky, 2018-06-06:
@poisonnuke , yes it is difficult and time-consuming to find out the different switches. Anyways good luck with your project.

closed 2018-06-06 by robotrovsky

poisonnuke, 2018-06-16:
ok Im back on this issue, but still trying to figure out, what is what.

Do I see it right, in data center, when you performed a config change (click on off for example), then a OUT group was created with data0 as the control message sent to the trackpad? If yes, is the "53" the bmRequest Type? Im wondering, because I cannot find the appropriate "lenght" field regarding to chapter 9 of the USB specification.

robotrovsky, 2018-06-18:
Hi @poisonnuke,
i recorded the data some time ago and dont remember very well. I think it might have been force touch on and off in the settings. But if you have a special use case, which you want to be sniffed, you can tell me what to do, and i will record the action.

poisonnuke, 2018-06-18:
I would like to take full control over the haptic feedback. Which means disable forcetouch in the first place (is this possible? tried this a few minutes ago with a mac pro of a colleague with no success) and then send haptic feedback impulse on my regard. If I could adjust the impulse-pattern and intensity, that would be quite nice too.

Ponyfleisch already achieved this for bluetooth, but as you mentioned, the requestIDs are different for USB and BT, and I need USB.

I ask myself, if the so called "hostClick" happens if one changes the intensity of the feedback or threshold or enables double Force Touch?

poisonnuke, 2018-06-22:
hm, could you tell me for the existing data, if I am right or wrong about the "IN" and "OUT" groups? So should I look into an "OUT" group for anything sent to the magic trackpad?

If yes, why am I not able to determine the "lenght" field in the request? Should be byte 8, but thats a way to large number there in all OUT packets.

thank you in advance