mwyborski/Linux-Magic-Trackpad-2-Driver #6: Can't get it to work on Fedora?

LeoUfimtsev, 2018-06-08:
Hello, 

I'm running Fedora 28 and I can't get it to work. (I have not tried earlier fedora versions).

I.e,
0) I build via makefile
1) I installed via ./install script.
2) after reboot, right click/scrolling doesn't work.
3) Following troubleshooting I get errors:
[08:05:44 hid]$ sudo rmmod hid_magicmouse
rmmod: ERROR: Module hid_magicmouse is builtin.
[08:05:58 hid]$ sudo insmod ./hid-magicmouse.ko
insmod: ERROR: could not insert module ./hid-magicmouse.ko: Invalid module format

Install log:
----------------------------------------------------------
sudo ./install.sh 
+ dkms_name=hid-magicmouse-dkms
+ dkms_version=4.17+magictrackpad2
+++ dirname ./install.sh
++ cd .
++ pwd
+ DIR=/home/lufimtse/git/Linux-Magic-Trackpad-2-Driver/scripts
+ dkms status -m hid-magicmouse-dkms -v 4.17+magictrackpad2
+ egrep '(added|built|installed)'
+ dkms add /home/lufimtse/git/Linux-Magic-Trackpad-2-Driver/scripts/../linux/drivers/hid

Creating symlink /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/source ->
                 /usr/src/hid-magicmouse-dkms-4.17+magictrackpad2

DKMS: add completed.
+ dkms status -m hid-magicmouse-dkms -v 4.17+magictrackpad2
+ egrep '(built|installed)'
+ dkms build hid-magicmouse-dkms/4.17+magictrackpad2

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=4.16.5-300.fc28.x86_64 KERNEL_VERSION=4.16.5-300.fc28.x86_64 all...
cleaning build area...

DKMS: build completed.
+ dkms status -m hid-magicmouse-dkms -v 4.17+magictrackpad2
+ egrep '(installed)'
+ dkms install --force hid-magicmouse-dkms/4.17+magictrackpad2

hid-magicmouse.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.16.5-300.fc28.x86_64/extra/
Adding any weak-modules

/etc/sysconfig/kernel: line 7: blacklist: command not found
depmod....

Backing up initramfs-4.16.5-300.fc28.x86_64.img to /boot/initramfs-4.16.5-300.fc28.x86_64.img.old-dkms
Making new initramfs-4.16.5-300.fc28.x86_64.img
(If next boot fails, revert to initramfs-4.16.5-300.fc28.x86_64.img.old-dkms image)
dracut........

DKMS: install completed.
--------------------------------------------------------------

x11 event log:  (at  181.646, I unplug & plug mouse back in).

tail -f ~/.local/share/xorg/Xorg.0.log
[    61.044] (II) event4  - Apple Inc. Magic Trackpad 2: device is a pointer
[    61.044] (II) event4  - Apple Inc. Magic Trackpad 2: device removed
[    61.044] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.1/0003:05AC:0265.000F/input/input20/event4"
[    61.044] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: MOUSE, id 13)
[    61.044] (**) Option "AccelerationScheme" "none"
[    61.045] (**) Apple Inc. Magic Trackpad 2: (accel) selected scheme none/0
[    61.045] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[    61.045] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[    61.045] (II) event4  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Mouse
[    61.045] (II) event4  - Apple Inc. Magic Trackpad 2: device is a pointer
[   177.318] (II) config/udev: removing device Apple Inc. Magic Trackpad 2
[   177.318] (**) Option "fd" "61"
[   177.318] (II) event4  - Apple Inc. Magic Trackpad 2: device removed
[   177.322] (II) UnloadModule: "libinput"
[   177.322] (II) systemd-logind: releasing fd for 13:68
[   181.524] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/mouse0)
[   181.524] (II) No input driver specified, ignoring this device.
[   181.524] (II) This device may have been added with another device file.
[   181.593] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event4)
[   181.593] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "evdev pointer catchall"
[   181.593] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput pointer catchall"
[   181.593] (II) Using input driver 'libinput' for 'Apple Inc. Magic Trackpad 2'
[   181.645] (II) systemd-logind: got fd for /dev/input/event4 13:68 fd 62 paused 0
[   181.645] (**) Apple Inc. Magic Trackpad 2: always reports core events
[   181.645] (**) Option "Device" "/dev/input/event4"
[   181.645] (**) Option "_source" "server/udev"
[   181.646] (II) event4  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Mouse
[   181.646] (II) event4  - Apple Inc. Magic Trackpad 2: device is a pointer
[   181.646] (II) event4  - Apple Inc. Magic Trackpad 2: device removed
[   181.646] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.1/0003:05AC:0265.0013/input/input21/event4"
[   181.646] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: MOUSE, id 13)
[   181.648] (**) Option "AccelerationScheme" "none"
[   181.650] (**) Apple Inc. Magic Trackpad 2: (accel) selected scheme none/0
[   181.650] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[   181.650] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[   181.651] (II) event4  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Mouse
[   181.651] (II) event4  - Apple Inc. Magic Trackpad 2: device is a pointer
q^C

robotrovsky, 2018-06-08:
@LeoUfimtsev someone in another discussion tested it with fedora and said:

> I can confirm this works on Fedora 28. You must recompile the kernel as by default the old hid_magicmouse driver is built in (not a module).

If you don't want to recompile the whole kernel, you could try to rename the module and load it with this new name. I think that might work.

LeoUfimtsev, 2018-06-08:
6 minute response time is impressive.

How do I rename the  module name?

Do I edit this guy:
Linux-Magic-Trackpad-2-Driver/linux/drivers/hid/dkms.conf

And change this to something like 'hid-magicmouse-custom' ?:
BUILT_MODULE_NAME[0]=hid-magicmouse

And then run the remove and then the install script again & reboot?

(do let me know if you know any articles that would be useful to read to better understand kernel modules, as I haven't worked much with them before).

robotrovsky, 2018-06-08:
@LeoUfimtsev the easiest way would be to go to linux/drivers/hid modify hid-magicmouse.c in line 696
and replace
`.name = "magicmouse",`
with
`.name = "magicmouse-custom",`

Then rename the whole file from hid-magicmouse.c to hid-magicmouse-custom.c.
And rename the two occurrences of "hid-magicmouse.o" to "hid-magicmouse-custom.o" in the makefile.
Then:
```
make
sudo insmod ./hid-magicmouse-custom.ko
tail -f ~/.local/share/xorg/Xorg.0.log
```

Now unplug the trackpad and plug it back in, to see which driver gets loaded.

closed 2018-06-08 by robotrovsky

LeoUfimtsev, 2018-06-08:
This worked, thanks for the instructions!

I'd be nice to get this into fedora somehow. I'll submit separate ticket.