mwyborski/Linux-Magic-Trackpad-2-Driver #7: how to get driver into Fedora?

LeoUfimtsev, 2018-06-08:
Hello, 

I got the trackpad to work on Fedora 28 (wirelessly & wired) with this driver by renaming it as per ticket #6 .

Considering I work as Software Engineer at Red Hat, I was wondering if we can somehow get this driver into upstream kernel and/or fedora. Perhaps I could nudge our guys to include this driver?

Have you looked into this, do you have any suggestions? Should I submit a fedora kernel report and ask them to take a look at this driver?

robotrovsky, 2018-06-08:
@LeoUfimtsev i will have to clean up the code. Then i will send it to the linux-input mailinglist. But if you have someone at red hat or fedora, whom i could include when i send in the patch, it would be great to get some support. My schedule is to submit the patch until end of this month. I don't want to hurry, and maybe i will change some details before submitting.

It is great that you and others are testing the driver!

mitermayer, 2019-04-11:
Whats the current state on having this driver on fedora ???

robotrovsky, 2019-04-11:
@mitermayer if you have a kernel >= 4.20 then you have the driver. But you will need a quirks file like described in the readme.

dos1, 2019-04-12:
After https://gitlab.freedesktop.org/libinput/libinput/merge_requests/214/diffs and https://github.com/systemd/systemd/pull/12282 reach the distros, everything should work out-of-the-box via either Bluetooth or USB.

robotrovsky, 2019-04-12:
@dos1 that is great to hear!!

closed 2019-06-28 by robotrovsky