mwyborski/Linux-Magic-Trackpad-2-Driver #9: Driver not loading for bluetooth connection (Ubuntu 18.04)

MeanderingCode, 2018-07-03:
The driver is not loaded when connecting the Magic Trackpad 2 via bluetooth in Ubuntu 18.04.

```
Jul 03 08:54:59 hostname kernel: hid-generic 0005:004C:0265.0096: unknown main item tag 0x0
Jul 03 08:54:59 hostname kernel: input: Magic Trackpad 2 as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256/0005:004C:0265.0096/input/input131
Jul 03 08:54:59 hostname kernel: hid-generic 0005:004C:0265.0096: input,hidraw4: BLUETOOTH HID v0.70 Mouse [Magic Trackpad 2] on e0:9d:31:da:51:5a
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) config/udev: Adding input device Magic Trackpad 2 (/dev/input/mouse2)
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) No input driver specified, ignoring this device.
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) This device may have been added with another device file.
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) config/udev: Adding input device Magic Trackpad 2 (/dev/input/event9)
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Magic Trackpad 2: Applying InputClass "libinput pointer catchall"
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) Using input driver 'libinput' for 'Magic Trackpad 2'
Jul 03 08:54:59 hostname upowerd[1327]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256/0005:004C:0265.0096
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) systemd-logind: got fd for /dev/input/event9 13:73 fd 98 paused 0
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Magic Trackpad 2: always reports core events
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Option "Device" "/dev/input/event9"
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Option "_source" "server/udev"
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) event9  - Magic Trackpad 2: is tagged by udev as: Mouse
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) event9  - Magic Trackpad 2: device is a pointer
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) event9  - Magic Trackpad 2: device removed
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256/0005:004C:0265.0096/input/input131/event9"
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) XINPUT: Adding extended input device "Magic Trackpad 2" (type: MOUSE, id 11)
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Option "AccelerationScheme" "none"
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Magic Trackpad 2: (accel) selected scheme none/0
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Magic Trackpad 2: (accel) acceleration factor: 2.000
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Magic Trackpad 2: (accel) acceleration threshold: 4
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) event9  - Magic Trackpad 2: is tagged by udev as: Mouse
Jul 03 08:54:59 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) event9  - Magic Trackpad 2: device is a pointer
```

I can unbind from `hid-generic` and bind to `magicmouse` like so:
```
sudo /bin/bash -c "echo 0005:004C:0265.0096 > /sys/bus/hid/drivers/hid-generic/unbind"
sudo /bin/bash -c "echo 0005:004C:0265.0096 > /sys/bus/hid/drivers/magicmouse//bind"
```
```
Jul 03 08:55:17 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) config/udev: removing device Magic Trackpad 2
Jul 03 08:55:17 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Option "fd" "98"
Jul 03 08:55:17 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) event9  - Magic Trackpad 2: device removed
Jul 03 08:55:17 hostname gnome-shell[32563]: g_array_unref: assertion 'array' failed
Jul 03 08:55:17 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) UnloadModule: "libinput"
Jul 03 08:55:17 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) systemd-logind: releasing fd for 13:73
Jul 03 08:55:17 hostname sudo[3295]: pam_unix(sudo:session): session closed for user root
Jul 03 08:55:17 hostname upowerd[1327]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256/0005:004C:0265.0096
Jul 03 08:55:17 hostname kernel: magicmouse 0005:004C:0265.0096: unknown main item tag 0x0
Jul 03 08:55:18 hostname kernel: input: Apple Inc. Magic Trackpad 2 as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256/0005:004C:0265.0096/input/input132
Jul 03 08:55:18 hostname kernel: magicmouse 0005:004C:0265.0096: input,hidraw4: BLUETOOTH HID v0.70 Mouse [Magic Trackpad 2] on e0:9d:31:da:51:5a
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/mouse2)
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) No input driver specified, ignoring this device.
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) This device may have been added with another device file.
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event9)
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput pointer catchall"
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput touchpad catchall"
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) Using input driver 'libinput' for 'Apple Inc. Magic Trackpad 2'
Jul 03 08:55:18 hostname upowerd[1327]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256/0005:004C:0265.0096
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) systemd-logind: got fd for /dev/input/event9 13:73 fd 98 paused 0
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Apple Inc. Magic Trackpad 2: always reports core events
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Option "Device" "/dev/input/event9"
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Option "_source" "server/udev"
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) event9  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Mouse Touchpad
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) event9  - Apple Inc. Magic Trackpad 2: device is a touchpad
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) event9  - Apple Inc. Magic Trackpad 2: device removed
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256/0005:004C:0265.0096/input/input132/event9"
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: TOUCHPAD, id 11)
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Option "AccelerationScheme" "none"
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Apple Inc. Magic Trackpad 2: (accel) selected scheme none/0
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) event9  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Mouse Touchpad
Jul 03 08:55:18 hostname /usr/lib/gdm3/gdm-x-session[1640]: (II) event9  - Apple Inc. Magic Trackpad 2: device is a touchpad
```

And the device works as expected with basic multitouch functionality.

Is this a failure to match IDs from the `linux/drivers/hid/hid-ids.h` file? I don't have much experience with input drivers.

igor-kupczynski, 2018-09-05:
I wonder if this is because of a kernel version. Ubuntu 18.04 has 4.15, but they seem to require 4.18 based on the readme.

StrangeNoises, 2018-09-05:
FWIW it's working on this Ubuntu 18.04... with the Mainline 4.18 kernel. Though I must say I didn't attempt it before upgrading the kernel because of that readme. ;-)

regis-leray, 2018-10-01:
yeah i succeed to make it work with ubuntu 18.04 and upgrade the kernel `4.18.10`

regis-leray, 2018-10-02:
did someone succeed to make it work with bluetooth ? i was only able to make it work with USB.
i cannot see the device with `bluetoothctl`

my bluetooth module works since im able to pair with my phone and  apple magic keyboard 2

StrangeNoises, 2018-10-02:
it worked with bluetooth for me, without me having to try, but remember it does require Bluetooth 4.0, so if the machine you're trying it on doesn't have that, it probably won't.

StrangeNoises, 2018-10-02:
I'm afraid I don't know. I belatedly just noticed your lsusb output showing that yours already is a Bluetooth 4.0 interface, so I'm out of ideas (unless, eg: you have two bluetooth interfaces and they're fighting?) It just worked for me so I didn't have to go crawling through the logs.

... and now I've sold the magic trackpad 2 anyway, for Reasons, so I can't try again.

FWIW:

When I used the trackpad in macOS it was mostly on a hackintosh, and there I used -- (plugs it in) -- as it turns out, exactly the same bluetooth adapter as you're using. So there shouldn't be any problem with the device itself being incompatible.

When I used it on this linux laptop - its bluetooth adapter is part of the Qualcomm Atheros wifi/bluetooth combined device, although it just shows up in lsusb as "Foxconn / Hon Hai", not very usefully.

regis-leray, 2018-10-02:
i was dealing with this issue also
https://github.com/robotrovsky/Linux-Magic-Trackpad-2-Driver/issues/14
i endup by creating  `/etc/rc.local` script which reload the driver

regis-leray, 2018-10-10:
oki i finally found the problem, on the magic trackpad i didnt switch on/off the trackpad.
I also install the latest kernel driver `4.18.13`

closed 2019-11-18 by robotrovsky