mwyborski/Linux-Magic-Trackpad-2-Driver #11: Can click but not move

nmattia, 2018-07-07:
I've managed to compile and load the module, but upon Blueeoth connection I can only click through the trackpad, not move the cursor. 

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.4 LTS
Release:	16.04
Codename:	xenial
$ uname -r
4.4.0-130-generic
```

The module appears to be loaded:

```
$ dkms status | head -n 1
hid-magicmouse-dkms, 4.17+magictrackpad2, 4.4.0-130-generic, x86_64: installed
$ lsmod | grep magic
hid_magicmouse         20480  0
hid                   122880  6 hidp,i2c_hid,hid_multitouch,hid_generic,usbhid,hid_magicmouse
```

This is what `/var/log/Xorg.0.log` shows when connecting the device:

```
[  1381.594] (II) config/udev: Adding input device Magic Trackpad 2 (/dev/input/mouse4)
[  1381.594] (II) No input driver specified, ignoring this device.
[  1381.594] (II) This device may have been added with another device file.
[  1381.658] (II) config/udev: Adding input device Magic Trackpad 2 (/dev/input/event23)
[  1381.658] (**) Magic Trackpad 2: Applying InputClass "evdev pointer catchall"
[  1381.658] (**) Magic Trackpad 2: Applying InputClass "Touchpads"
[  1381.658] (II) Using input driver 'mtrack' for 'Magic Trackpad 2'
[  1381.658] (**) Magic Trackpad 2: always reports core events
[  1381.658] (**) Magic Trackpad 2: always reports core events
[  1381.658] (**) Option "FingerHigh" "10"
[  1381.658] (**) Option "FingerLow" "10"
[  1381.658] (**) Option "ButtonIntegrated" "true"
[  1381.658] (**) Option "TapButton1" "1"
[  1381.658] (**) Option "TapButton2" "3"
[  1381.658] (**) Option "TapButton3" "2"
[  1381.658] (**) Option "TapButton4" "0"
[  1381.658] (**) Option "ClickTime" "25"
[  1381.658] (**) Option "ScrollDistance" "75"
[  1381.658] (**) Option "ScrollUpButton" "5"
[  1381.658] (**) Option "ScrollDownButton" "4"
[  1381.658] (**) Option "ScrollLeftButton" "7"
[  1381.658] (**) Option "ScrollRightButton" "6"
[  1381.659] (**) Option "SwipeDistance" "1"
[  1381.659] (**) Option "SwipeUpButton" "1"
[  1381.659] (**) Option "SwipeDownButton" "1"
[  1381.659] (**) Option "SwipeLeftButton" "1"
[  1381.659] (**) Option "SwipeRightButton" "1"
[  1381.659] (**) Option "TapDragEnable" "false"
[  1381.659] (**) Option "Sensitivity" "0.55"
[  1381.659] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:4/0005:004C:0265.000D/input/input32/event23"
[  1381.659] (II) XINPUT: Adding extended input device "Magic Trackpad 2" (type: TOUCHPAD, id 21)
[  1381.659] (II) device control: init
[  1381.659] (**) Option "Device" "/dev/input/event23"
[  1381.659] (II) mtrack: devname: Magic Trackpad 2
[  1381.659] (II) mtrack: devid: 4c 265 70
[  1381.659] (II) mtrack: caps: left right
[  1381.680] (**) Magic Trackpad 2: (accel) keeping acceleration scheme 1
[  1381.680] (**) Magic Trackpad 2: (accel) acceleration profile 0
[  1381.680] (**) Magic Trackpad 2: (accel) acceleration factor: 2.000
[  1381.681] (**) Magic Trackpad 2: (accel) acceleration threshold: 4
[  1381.681] (II) device control: on
[  1381.681] (WW) Touchpad has minimal capabilities. Some features will be unavailable.
```

I am a complete noob in kernel/driver world, but those lines look suspicious:

```
[  1381.594] (II) config/udev: Adding input device Magic Trackpad 2 (/dev/input/mouse4)
[  1381.594] (II) No input driver specified, ignoring this device.
[  1381.594] (II) This device may have been added with another device file.
```

Could the cursor movements somehow go through `/dev/input/mouse4` for which the driver is not used?

Thanks in advance!

gzm55, 2018-07-31:
the same problem on both bluetooth and usb connection.

toszter, 2018-09-05:
Same problem here. But if I push /just/ hard enough, the mouse starts to twitch a little bit and even move from moment to moment. Weird.

laagland, 2018-10-13:
I'm having the same problem. It seems the sensitivity is way too low. I've tested with xinput and only when pressed hard enough I get motion detection. But the pressure borders on performing a click, so not very useful.

@robotrovsky Is there a way to adjust the sensitivity with these drivers?

Using Ubuntu 18.04.1 LTS with kernel 4.18.0-041800-generic

badaz, 2018-12-24:
@robotrovsky Hi, I'm having this sensitivity problem too when connected over bluetooth, I thought it would be fixed with the release of linux kernel 4.20 today and integration of the driver to the kernel (I uninstalled your driver before installing the new kernel) but the pb is still there and I'm not sure I understand the answer you provided :

> This has to be handled in userspace and i think as well, that it would be better.

Is there an easy way for the end user to adjust pressure sensitivity so that one can use a MT2 over bluetooth without having to apply so much pressure to make the cursor move ?

robotrovsky, 2018-12-24:
@badaz glad that you sorted it out so quick. I installed 4.20 today but my ordinary touchpad stopped working. In 2 weeks I have access to the MT2 and I will try it again. I want to add a working libinput quirks file to the repo.

badaz, 2018-12-24:
That would be nice!

leandromaniezo, 2019-04-08:
I'm using ubuntu 18.04 on a Sony Vaio VPC, the Magic Trackpad 2 works just with one finger, I'd like to set it to work at least two finger scrolling.

Have you tried xinput, libinput and interfaces, but without success, have any tips to help?

Thanks

robotrovsky, 2019-04-11:
@leandromaniezo 18.04 doesn't come with the driver. The driver is included from 4.20 onwards. I recommend you to wait a few days and upgrade to Ubuntu 19.04, there everything works out of the box.

closed 2019-04-11 by robotrovsky