mwyborski/Linux-Magic-Trackpad-2-Driver #12: Build error on Linux 4.18-rc4

alanorth, 2018-07-14:
Driver works great on Linux 4.17 (thanks!), but there's a build error on Linux 4.18-rc4. How is the work to upstream the changes to this driver going?

Here is DKMS install:

```
# dkms install -m hid-magicmouse-dkms -v 4.17+magictrackpad2
Kernel preparation unnecessary for this kernel.  Skipping...
Building module:
cleaning build area...
make -j8 KERNELRELEASE=4.18.0-rc4-mainline KERNEL_VERSION=4.18.0-rc4-mainline all...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.18.0-rc4-mainline (x86_64)
Consult /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/make.log for more information.
```

And here is the error log:

```
DKMS make.log for hid-magicmouse-dkms-4.17+magictrackpad2 for kernel 4.18.0-rc4-mainline (x86_64)
Sat Jul 14 14:08:04 EEST 2018
make -C /lib/modules/4.18.0-rc4-mainline/build M=/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build modules
make[1]: Entering directory '/usr/lib/modules/4.18.0-rc4-mainline/build'
  CC [M]  /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.o
/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c: In function ‘magicmouse_probe’:
/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:627:12: error: too few arguments to function ‘hid_register_report’
   report = hid_register_report(hdev, HID_INPUT_REPORT,
            ^~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:18:
./include/linux/hid.h:878:20: note: declared here
 struct hid_report *hid_register_report(struct hid_device *device,
                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:630:12: error: too few arguments to function ‘hid_register_report’
   report = hid_register_report(hdev, HID_INPUT_REPORT,
            ^~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:18:
./include/linux/hid.h:878:20: note: declared here
 struct hid_report *hid_register_report(struct hid_device *device,
                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:632:12: error: too few arguments to function ‘hid_register_report’
   report = hid_register_report(hdev, HID_INPUT_REPORT,
            ^~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:18:
./include/linux/hid.h:878:20: note: declared here
 struct hid_report *hid_register_report(struct hid_device *device,
                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:636:13: error: too few arguments to function ‘hid_register_report’
    report = hid_register_report(hdev, HID_INPUT_REPORT,
             ^~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:18:
./include/linux/hid.h:878:20: note: declared here
 struct hid_report *hid_register_report(struct hid_device *device,
                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:639:13: error: too few arguments to function ‘hid_register_report’
    report = hid_register_report(hdev, HID_INPUT_REPORT,
             ^~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:18:
./include/linux/hid.h:878:20: note: declared here
 struct hid_report *hid_register_report(struct hid_device *device,
                    ^~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:324: /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.o] Error 1
make[1]: *** [Makefile:1500: _module_/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.18.0-rc4-mainline/build'
make: *** [Makefile:9: all] Error 2
```

robotrovsky, 2018-07-24:
Thanks for reporting. I will look at this, before submitting the patch.

rohitpid, 2018-07-24:
from hid_magicmouse in mainline

> 
	if (id->product == USB_DEVICE_ID_APPLE_MAGICMOUSE)
		report = hid_register_report(hdev, HID_INPUT_REPORT,
			MOUSE_REPORT_ID, 0);
	else { /* USB_DEVICE_ID_APPLE_MAGICTRACKPAD */
		report = hid_register_report(hdev, HID_INPUT_REPORT,
			TRACKPAD_REPORT_ID, 0);

rohitpid, 2018-07-24:
https://github.com/robotrovsky/Linux-Magic-Trackpad-2-Driver/pull/15 here is a PR with the changes.

alanorth, 2018-07-24:
Working for me in Linux 4.18-rc6 on Arch Linux. Thanks @rohitpid!

closed 2018-07-24 by alanorth