mwyborski/Linux-Magic-Trackpad-2-Driver #14: I have to manually bind hid_magicmouse to the device. hid-generic steals it at boot.

rohitpid, 2018-07-24:
I have to run 
echo -n "0005:004C:0265.0004" > /sys/bus/hid/drivers/hid-generic/unbind
echo -n "0005:004C:0265.0004" > /sys/bus/hid/drivers/magicmouse/bind
 
to make this work. 0x265 is listed in the hid-ids.c so I do we need to blacklist it from hid-generic? Is this a race condition?

alanorth, 2018-07-24:
Are both the `hid_generic` and `hid_magicmouse` modules loaded? I simply `rmmod hid_generic; modprobe hid_magicmouse` instead of doing the unbind/bind.

rohitpid, 2018-07-24:
I need hid_generic loaded for my keyboard and the magicmouse2 (which I would like to port over to use hid_magicmouse but it isn't currently supported).

satur9nine, 2018-08-02:
I have the same problem on linux kernel 4.13 (i ported this driver and loaded it via dkms), hid-generic is chosen as the driver but I can fix it with the above mentioned bind/unbind commands. I haven't figured why, it seems that linux should be finding and preferring this driver via MODULE_DEVICE_TABLE and/or id_table in hid-magicmouse.c but it's not working?

rohitpid, 2018-08-02:
@satur9nine I haven't been having this issue on 4.18 since updating. Try it out with 4.18 if you can. If it persists on newer kernels, I'll try to debug further and find a solution.

satur9nine, 2018-08-03:
I see now that kernel 4.13 would require a more complex solution because there is a list of devices in hid-core.c which controls which driver gets used.

regis-leray, 2018-10-01:
im facing the same issue
with ubuntu `18.04` with kernel `4.18.10`
i declared `hid_magicmouse`

```
cp ~/sources/Linux-Magic-Trackpad-2-Driver-master/linux/drivers/hid/hid-magicmouse.ko /lib/modules/4.18.10-041810-generic/kernel/drivers/
echo 'hid-magicmouse' | sudo tee -a /etc/modules
sudo depmod
```

at the startup i can see the module loaded
```
lsmod  | grep hid
hid_magicmouse         20480  0
mac_hid                16384  0
hid_generic            16384  0
usbhid                 49152  0
hid                   126976  3 usbhid,hid_generic,hid_magicmouse
```

leandromaniezo, 2019-04-08:
I'm using ubuntu 18.04 on a Sony Vaio VPC, the Magic Trackpad 2 works just with one finger, I'd like to set it to work at least two finger scrolling.

Have you tried xinput, libinput and interfaces, but without success, have any tips to help?

Thanks

mani47, 2019-05-27:
> im facing the same issue
> with ubuntu `18.04` with kernel `4.18.10`
> i declared `hid_magicmouse`
> 
> ```
> cp ~/sources/Linux-Magic-Trackpad-2-Driver-master/linux/drivers/hid/hid-magicmouse.ko /lib/modules/4.18.10-041810-generic/kernel/drivers/
> echo 'hid-magicmouse' | sudo tee -a /etc/modules
> sudo depmod
> ```
> 
> at the startup i can see the module loaded
> 
> ```
> lsmod  | grep hid
> hid_magicmouse         20480  0
> mac_hid                16384  0
> hid_generic            16384  0
> usbhid                 49152  0
> hid                   126976  3 usbhid,hid_generic,hid_magicmouse
> ```
> 
> and this what i do to fixit
> 
> ```
> sudo rmmod hid_magicmouse
> sudo insmod ./hid-magicmouse.ko 
> ```

Have you signed the hid_magicmouse module?
```
$ dmesg | grep magic
[    3.483318] hid_magicmouse: loading out-of-tree module taints kernel.
[    3.483335] hid_magicmouse: module verification failed: signature and/or required key missing - tainting kernel
[   64.695904] magicmouse 0005:004C:0269.0005: unknown main item tag 0x0
[   64.696024] magicmouse 0005:004C:0269.0005: input,hidraw4: BLUETOOTH HID v0.70 Mouse [Mani Malek Mohammadi’s Mouse] on 14:20:5e:06:f4:a5
```

dos1, 2019-05-28:
@regis-leray's and @leandromaniezo's issues seem like unrelated ones. This one (having to bind hid_magicmouse to the device), as far as I understand it, it a correct and expected behavior on kernels older than 4.18, unless you rebuild your whole kernel with modified `hid-quirks.c` file.

jeremysf, 2019-07-18:
I'm on 19.04 and I can initially get magic mouse 2 with scrolling to work, but it will not survive reboots. When I look, it is bound to hid_generic, but no amount of rmmod/insmod works for me (I can run the uninstall script and reboot, and then reinstall and it works until I reboot).

I am able to unbind it from the hid_generic, but if I try to bind it to the magicmouse driver, I just get "can't find device" type messages :/

brainard52, 2019-12-02:
> Have you signed the hid_magicmouse module?
> 
> ```
> $ dmesg | grep magic
> [    3.483335] hid_magicmouse: module verification failed: signature and/or required key missing - tainting kernel

Is signing hid_magicmouse a complex process? I'm a bit of a neophyte when it comes to kernel modules. This is on a fresh installation of Ubuntu 19.10.