mwyborski/Linux-Magic-Trackpad-2-Driver #16: Makefile failing

cam-carter, 2018-07-25:
Here's the output of `make.log`

I honestly have no idea what's going on.  This is happening with the dkms scripts when running `sudo ./install.sh` inside `./scripts`, as well as, just running `make` inside `./linux/drivers/hid`.  Any help would be appreciated.

```
DKMS make.log for hid-magicmouse-dkms-4.17+magictrackpad2 for kernel 4.17.9-041709-generic (x86_64)
Tue Jul 24 20:07:23 EDT 2018
make -C /lib/modules/4.17.9-041709-generic/build M=/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.17.9-041709-generic'
  CC [M]  /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.o
/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c: In function ‘magicmouse_probe’:
/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:636:12: error: too many arguments to function ‘hid_register_report’
   report = hid_register_report(hdev, HID_INPUT_REPORT,
            ^~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:18:0:
./include/linux/hid.h:868:20: note: declared here
 struct hid_report *hid_register_report(struct hid_device *device, unsigned type, unsigned id);
                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:639:12: error: too many arguments to function ‘hid_register_report’
   report = hid_register_report(hdev, HID_INPUT_REPORT,
            ^~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:18:0:
./include/linux/hid.h:868:20: note: declared here
 struct hid_report *hid_register_report(struct hid_device *device, unsigned type, unsigned id);
                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:641:12: error: too many arguments to function ‘hid_register_report’
   report = hid_register_report(hdev, HID_INPUT_REPORT,
            ^~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:18:0:
./include/linux/hid.h:868:20: note: declared here
 struct hid_report *hid_register_report(struct hid_device *device, unsigned type, unsigned id);
                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:645:13: error: too many arguments to function ‘hid_register_report’
    report = hid_register_report(hdev, HID_INPUT_REPORT,
             ^~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:18:0:
./include/linux/hid.h:868:20: note: declared here
 struct hid_report *hid_register_report(struct hid_device *device, unsigned type, unsigned id);
                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:648:13: error: too many arguments to function ‘hid_register_report’
    report = hid_register_report(hdev, HID_INPUT_REPORT,
             ^~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.c:18:0:
./include/linux/hid.h:868:20: note: declared here
 struct hid_report *hid_register_report(struct hid_device *device, unsigned type, unsigned id);
                    ^~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:318: recipe for target '/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.o' failed
make[2]: *** [/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build/hid-magicmouse.o] Error 1
Makefile:1577: recipe for target '_module_/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build' failed
make[1]: *** [_module_/var/lib/dkms/hid-magicmouse-dkms/4.17+magictrackpad2/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.17.9-041709-generic'
Makefile:9: recipe for target 'all' failed
make: *** [all] Error 2
```

adam-h, 2018-07-25:
@cam-carter You need to revert ecfde9fce81f189a25da5a3f62db68d8272d37fb as I mentioned [here](https://github.com/robotrovsky/Linux-Magic-Trackpad-2-Driver/commit/ecfde9fce81f189a25da5a3f62db68d8272d37fb#commitcomment-29823051) for 4.17 support. Then it'll work :)

cam-carter, 2018-07-25:
@adam-h everything's working great now!  Thanks a lot.  Work hooked me up with the trackpad, but I'm the only GNU/Linux user in the office.  Didn't want to feel left out haha

Will this work for older kernel versions as well, or just 4.17?  I was initially trying with 4.15.

adam-h, 2018-07-25:
I've not tried, I would guess not without modifications.

Glad that worked for you, it is a very nice Trackpad

cam-carter, 2018-07-25:
I've got the install working with an older kernel version.  It just took the same revert.  Now I've installed xf86-input-mtrack and it's not playing nicely with the trackpad.

The followed is the output of `tail -f ~/.local/share/xorg/Xorg.0.log`:
```
[  1106.371] (**) Option "Sensitivity" "0.55"
[  1106.371] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4:1.0/bluetooth/hci0/hci0:12/0005:004C:0265.0004/input/input21/event18"
[  1106.371] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: TOUCHPAD, id 17)
[  1106.371] (II) mtrack[11] driver/mtrack.c:564: (II) device control: init
[  1106.371] (**) Option "Device" "/dev/input/event18"
[  1106.371] (EE) xf86OpenSerial: Cannot open device /dev/input/event18
	Permission denied.
[  1106.371] (EE) mtrack[12] driver/mtrack.c:148: (EE) cannot open device Apple Inc. Magic Trackpad 2
[  1106.371] (EE) Couldn't init device "Apple Inc. Magic Trackpad 2"
[  1106.372] (II) UnloadModule: "mtrack"
```

closed 2019-06-28 by robotrovsky