mwyborski/Linux-Magic-Trackpad-2-Driver #25: Have to manually run sudo insmod ./hid-magicmouse.ko after each restart

x-ji, 2018-10-13:
Sorry if this question is noobish. I'm not familiar with dkms. I ran the `install.sh` script, and it seems that `dkms` does install a module called `4.18+magictrackpad2`. However, after each restart I can only perform single click with no other actions available.

If I run `sudo rmmod hid_magicmouse`, apparently the module is not currently loaded. After I run `sudo insmod ./hid-magicmouse.ko`, the magic trackpad works correctly (I still don't exactly know how to configure it though. The settings manager provided by the WM doesn't seem to work. Should I manually write an XOrg `.conf` file?).

The output of `dkms status` is as such:

`hid-magicmouse-dkms, 4.18+magictrackpad2, 4.18.12-1-MANJARO, x86_64: installed`

I wonder why `hid-magicmouse-dkms` and `4.18+magictrackpad2` exist at the same time. It seems that I didn't do something correctly?

scottnuma, 2018-10-14:
Can you post the result of `tail -f ~/.local/share/xorg/Xorg.0.log`? Both before and after you run `sudo insmod ./hid-magicmouse.ko`

x-ji, 2018-10-14:
I tried to follow the debug guide before but couldn't find the log file. After a bit digging I found the file at `/var/log/Xorg.0.log`.

Before executing the command:

```
[     8.579] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 Mouse (/dev/input/event4)
[     8.579] (**) Apple Inc. Magic Trackpad 2 Mouse: Applying InputClass "evdev pointer catchall"
[     8.579] (**) Apple Inc. Magic Trackpad 2 Mouse: Applying InputClass "libinput pointer catchall"
[     8.579] (II) Using input driver 'libinput' for 'Apple Inc. Magic Trackpad 2 Mouse'
[     8.579] (**) Apple Inc. Magic Trackpad 2 Mouse: always reports core events
[     8.579] (**) Option "Device" "/dev/input/event4"
[     8.579] (**) Option "_source" "server/udev"
[     8.636] (II) event4  - Apple Inc. Magic Trackpad 2 Mouse: is tagged by udev as: Mouse
[     8.636] (II) event4  - Apple Inc. Magic Trackpad 2 Mouse: device is a pointer
[     8.637] (II) event4  - Apple Inc. Magic Trackpad 2 Mouse: device removed
[     8.693] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:05AC:0265.0002/input/input5/event4"
[     8.693] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2 Mouse" (type: MOUSE, id 9)
[     8.694] (**) Option "AccelerationScheme" "none"
[     8.694] (**) Apple Inc. Magic Trackpad 2 Mouse: (accel) selected scheme none/0
[     8.694] (**) Apple Inc. Magic Trackpad 2 Mouse: (accel) acceleration factor: 2.000
[     8.694] (**) Apple Inc. Magic Trackpad 2 Mouse: (accel) acceleration threshold: 4
[     8.754] (II) event4  - Apple Inc. Magic Trackpad 2 Mouse: is tagged by udev as: Mouse
[     8.754] (II) event4  - Apple Inc. Magic Trackpad 2 Mouse: device is a pointer
[     8.756] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 Mouse (/dev/input/mouse0)
[     8.756] (II) No input driver specified, ignoring this device.
[     8.756] (II) This device may have been added with another device file.
```

After:

```
[   553.995] (II) config/udev: removing device Apple Inc. Magic Trackpad 2 Mouse
[   553.995] (II) event4  - Apple Inc. Magic Trackpad 2 Mouse: device removed
[   554.021] (II) UnloadModule: "libinput"
[   554.040] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/mouse0)
[   554.040] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "touchpad"
[   554.040] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "15"
[   554.040] (II) Using input driver 'libinput' for 'Apple Inc. Magic Trackpad 2'
[   554.040] (**) Apple Inc. Magic Trackpad 2: always reports core events
[   554.040] (**) Option "Device" "/dev/input/mouse0"
[   554.040] (**) Option "_source" "server/udev"
[   554.161] (II) mouse0  - failed to create input device '/dev/input/mouse0'.
[   554.161] (EE) libinput: Apple Inc. Magic Trackpad 2: Failed to create a device for /dev/input/mouse0
[   554.161] (EE) PreInit returned 2 for "Apple Inc. Magic Trackpad 2"
[   554.161] (II) UnloadModule: "libinput"
[   554.323] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event4)
[   554.324] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "evdev touchpad catchall"
[   554.324] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput touchpad catchall"
[   554.324] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "touchpad"
[   554.324] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "15"
[   554.324] (II) Using input driver 'libinput' for 'Apple Inc. Magic Trackpad 2'
[   554.324] (**) Apple Inc. Magic Trackpad 2: always reports core events
[   554.324] (**) Option "Device" "/dev/input/event4"
[   554.324] (**) Option "_source" "server/udev"
[   554.383] (II) event4  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Touchpad
[   554.385] (II) event4  - Apple Inc. Magic Trackpad 2: device is a touchpad
[   554.385] (II) event4  - Apple Inc. Magic Trackpad 2: device removed
[   554.440] (**) Option "Tapping" "on"
[   554.440] (**) Option "AccelSpeed" "0.8"
[   554.440] (**) Option "NaturalScrolling" "true"
[   554.441] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:05AC:0265.0002/input/input23/event4"
[   554.441] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: TOUCHPAD, id 9)
[   554.444] (**) Option "AccelerationScheme" "none"
[   554.445] (**) Apple Inc. Magic Trackpad 2: (accel) selected scheme none/0
[   554.445] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[   554.445] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[   554.503] (II) event4  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Touchpad
[   554.505] (II) event4  - Apple Inc. Magic Trackpad 2: device is a touchpad
```

So it seems that the driver wasn't installed correctly after all and the device wasn't recognized at first? But when I try `sudo dkms install hid-magicmouse-dkms/4.18+magictrackpad2` I get:
"Module hid-magicmouse-dkms/4.18+magictrackpad2 already installed on kernel 4.18.12-1-MANJARO/x86_64"

alanorth, 2018-10-14:
Since you're on Manjaro (aka Arch) I'm guessing you're having problems because your distro's kernel already has a `hid-magicmouse` module that gets loaded when you do `modprobe`. My Arch has:

```
# find /usr/ -iname "*magicmouse.ko*"
/usr/lib/modules/4.18.12-arch1-1-ARCH/kernel/drivers/hid/hid-magicmouse.ko.xz
/usr/lib/modules/4.18.12-arch1-1-ARCH/kernel/drivers/hid/hid-magicmouse.ko
```

The `hid-magicmouse.ko.xz` comes from the distro kernel, and the `hid-magicmouse.ko` comes from DKMS. Try manually insmodding the `hid-magicmouse.ko` module. Personally, I always remove the one from the distro every time the kernel updates. It will be nice when this module is in the upstream kernel so we don't have to do that anymore!

x-ji, 2018-10-15:
@alanorth Thanks. This seems to indeed be the reason. After manually deleting the `.ko.xz` file that was originally present in the system, and reintsalling the driver, it seems to work immediately from a fresh boot. I was also encountering system freezes related to the driver. Hopefully this will solve that problem as well.

closed 2018-10-15 by x-ji