mwyborski/Linux-Magic-Trackpad-2-Driver #26: 4.18.12: gen-1 magicmouse acts only as 2-button mouse

rburcham, 2018-11-13:
I've documented details of my observations here:  https://forums.gentoo.org/viewtopic-t-1089352.html

In essence I cannot get the gen-1 magicmouse to middle click or v-scroll or h-scroll.

The hid_magicmouse module discovers and drives the device according to dmesg.  Running xinput with list, list-props and set-props all act as you would expect, but the device never behaves differently, that is enabling the middle button or scroll wheel props is successful, but the device continues to behave as a 2-button mouse.  Running evtest never shows any middle click or h-scroll or v-scroll activity.

This device used to operate properly but I'm unsure what's different about the driver now.

I noticed in the file /usr/share/libinput/50-system-apple.quirks

```
# The Apple MagicMouse has a touchpad built-in but the kernel still
# emulates a full 2/3 button mouse for us. Ignore anything from the
# ABS interface
[Apple MagicMouse]
MatchUdevType=mouse
MatchBus=bluetooth
MatchVendor=0x05AC
MatchProduct=0x030D
AttrEventCodeDisable=EV_ABS
```

I don't know how to interpret the comment... What does ABS mean?

robotrovsky, 2018-11-14:
@rburcham , sorry to hear, that this somehow broke your working device. How did you install the new driver? Can you give me a link to the package? The kernel 4.18 does not come with this driver (neither 4.19, but 4.20), so either you have it installed manually, or this is not a driver related issue. And how did you connect? Via bluetooth or USB? Do you have the same effect, independent of the interface? I think to track this down, you should try 2 things:

1. Make sure, that you really have the new driver installed. Grep for "TRACKPAD2_USB_REPORT_ID" in linux-source/drivers/hid/hid-magicmouse.c
2. Use the old driver (before the magicmouse2 patch)
3. Check if something with the libinput quirks-files changed.

rburcham, 2018-11-14:
Thanks for the reply @robotrovsky.  I am on gentoo, running a gentoo curated kernel 4.18.12 from the gentoo-sources package in portage.  I agree the new version of the driver likely is not in there:

`$ grep TRACKPAD2_USB_REPORT_ID drivers/hid/hid-magicmouse.c`

Further, hid_magicmouse is reporting the params:

```
# grep "" /sys/module/hid_magicmouse/parameters/*
/sys/module/hid_magicmouse/parameters/emulate_3button:Y
/sys/module/hid_magicmouse/parameters/emulate_scroll_wheel:Y
/sys/module/hid_magicmouse/parameters/report_undeciphered:N
/sys/module/hid_magicmouse/parameters/scroll_acceleration:N
/sys/module/hid_magicmouse/parameters/scroll_speed:32
```

As you say, I have to think I'm looking at the impact of some kind of change in libinput.  Perhaps a quirk change or a wholesale change in behavior.  I admit I don't know what version I was running on my original gentoo'd MBP 2012 where the magic mouse behaved flawlessly.

Since gentooing this Dell Precision 7330 I can see from emerge.log that within the past two weeks I have updated from `dev-libs/libinput-1.12.1-r1` to `dev-libs/libinput-1.12.2` to `dev-libs/libinput-1.12.3`, and the magic mouse has always behaved like a 2-button mouse across the board.

```
# emerge libinput -s
  
[ Results for search key : libinput ]
Searching...

*  dev-libs/libinput
      Latest version available: 1.12.3
      Latest version installed: 1.12.3
      Size of files: 508 KiB
      Homepage:      https://www.freedesktop.org/wiki/Software/libinput/
      Description:   Library to handle input devices in Wayland
      License:       MIT
... 

*  x11-drivers/xf86-input-libinput
      Latest version available: 0.28.1
      Latest version installed: 0.28.1
      Size of files: 368 KiB
      Homepage:      https://www.x.org/wiki/ https://cgit.freedesktop.org/
      Description:   X.org input driver based on libinput
      License:       MIT

*  x11-misc/libinput-gestures
      Latest version available: 2.39
      Latest version installed: [ Not Installed ]
      Size of files: 18 KiB
      Homepage:      https://github.com/bulletmark/libinput-gestures
      Description:   Actions gestures on your touchpad using libinput
      License:       GPL-3+

[ Applications found : 8 ]
```

I'll see if I can roll back the libinput.

robotrovsky, 2018-11-14:
You are welcome @rburcham. I think as well, that this is a libinput and no driver issue, so i will close this issue. Good luck ;-)

closed 2018-11-14 by robotrovsky

rburcham, 2018-11-30:
Sorry to bump this item @robotrovsky , but could you help me interpret this dmesg output when the hid_magicmouse discovers and drives the device:

```
[Fri Nov 16 15:30:27 2018] magicmouse 0005:05AC:030D.0005: unknown main item tag 0x0 
[Fri Nov 16 15:30:27 2018] input: rburcham’s Mouse as /devices/virtual/misc/uhid/0005:05AC:030D.0005/input/input24 
[Fri Nov 16 15:30:27 2018] magicmouse 0005:05AC:030D.0005: input,hidraw4: BLUETOOTH HID v3.06 Mouse [rburcham’s Mouse] on <redacted> 
```

Specifically what does the "unknown main item tag" mean?  And is the input,hidraw4 correct for a gen-1 magic mouse?

Is anyone able to actually drive their gen-1 magicmouse properly these days?  It really seems like there are a lot of folks making the same observations I'm making.

guyzmo, 2019-07-31:
@rburcham did you end up solving your issue? I opened a new issue :point_up: to track research and progress on the issue to maybe get to a solution, given @robotrovsky is ok with that.

rburcham, 2019-08-05:
Never did.  I did get a response on linux-input pointing the finger at the kernel module:

https://gitlab.freedesktop.org/libinput/libinput/issues/200#note_95204

But no joy.  Honestly I do feel that the commit (redesign) referenced in that thread must have something to do with the issue.  It's possible something was lost in translation, or maybe Peter didn't look closely enough, as the change referenced dealt with MT in the magicmouse kernel module which is essentially the problem.

I bought an MS arc mouse and have been using it ever since.