mwyborski/Linux-Magic-Trackpad-2-Driver #28: Any support for silent clicking?

andportnoy, 2018-11-20:
Apologies if this has been addressed somewhere else.

robotrovsky, 2018-11-21:
@andportnoy the driver only supports basic functionality. Neither Force-Touch nor configuration of click-strength is configurable right now. Maybe i (or someone else) will have a look at it in the future when 4.20 which includes the driver is released, but right now i am not working on the driver. You could use tap-to-click in the meantime.

andportnoy, 2018-11-21:
@robotrovsky Thank you for the clarification and thank you for the terrific work! I've been using the touchpad for a couple of days now and it's working great.

robotrovsky, 2018-11-21:
@andportnoy you are welcome. Glad to hear that people are using their trackpad.

andportnoy, 2018-11-21:
@robotrovsky Just wondering, how did you figure out how to build the driver and what the trackpad is sending? Was it through some kind of packet interception?

robotrovsky, 2018-11-21:
Yes, i started when i saw, that the Trackpad is working under Windows. So i used Wireshark to sniff the USB protocol. Once i figured out how to enable 10 Finger reporting i switched to Linux and compared the outcome of the driver with the Trackpad of my MacBook Touchpad. This was also why i first implemented it in the bcm5974 (touchpad driver). Afterwards i ported it over to the magicmouse driver which already contained the magic trackpad I, and added the bluetooth protocol, which @ponyfleisch reverse engineered (with OSX Bluetooth analyzer).

robotrovsky, 2018-11-21:
I think @ponyfleisch also has some code to configure the strength of the click. Have a look at it here:
https://github.com/ponyfleisch/hid-magictrackpad2

andportnoy, 2019-01-02:
@robotrovsky Thank you!

dos1, 2019-01-05:
Thanks to all the hints by @ponyfleisch and @robotrovsky and a little help from packet sniffer, I've figured it out! (at least for Bluetooth, no idea if it applies to USB as well yet)

The click strength can be configured without switching the device out of autonomous clicking mode (using `hid_hw_raw_request`) by using following message:

```
0xF2, 0x22/23, 0x01, 0x??, 0x78, 0x02, 0x??, 0x24, 0x30, 0x06, 0x01, 0x??, 0x18, 0x48, 0x13
```

Index 1 determines which feedback event you want to configure (0x22 is click, 0x23 is release).
Indexes 3, 6 and 11 (question marks) configure the feedback strength.

Values used by macOS:
- strength: low
  - click: 0x15, 0x04, 0x04
  - release: 0x10, 0x00, 0x00
- strength: medium
  - click: 0x17, 0x06, 0x06
  - release: 0x14, 0x00, 0x00
- strength: high
  - click: 0x1E, 0x08, 0x08
  - release: 0x18, 0x02, 0x02

"Silent mode" is the same, but two last values are always 0x00.

I've played a bit and seems that all three values can be set between 0x00 and 0xFF for various feedback configurations, so you don't have to be limited to what macOS lets you choose.

I'll try to prepare a proper patch and send it to the kernel maillist as soon as I figure out how to properly export those values as some config options :)

dos1, 2019-01-05:
I've just checked it on USB and the messages are the same (minus the first 0xF2 byte), only aren't sent as HID requests, but directly through URB_CONTROL.

dos1, 2019-01-05:
If someone wants to play with it, I have put my code at https://github.com/dos1/Linux-Magic-Trackpad-2-Driver for now :)

kirelagin, 2020-10-26:
Hmm, so what’s the status of this work? Are there plans to submit the patch for inclusion in the kernel?

dos1, 2020-10-26:
Hah, it somehow fell through cracks. I'll likely take a look at this driver in the next months since I've recently seen some kernel panics coming from it, so that may be a good occasion to at least send an RFC :)

kirelagin, 2020-10-26:
Ok, great, in the meantime, do you happen to know if there is any easy way to access the hidraw interface from the userspace? I mean, I was just thinking, in theory, it should be possible to somehow write this magic byte sequence without a kernel module?

dos1, 2020-10-26:
Yeah, I haven't really rushed with sending that patch out right away since I had a suspicion that it may be a thing that's better left for the user space. However, I haven't looked into hidraw at all and have no experience with it, so I don't know if there aren't some gotchas waiting to be discovered (especially given the fact that it has to be sent differently over USB and Bluetooth). The messages don't seem complicated though, so I think there's a chance it should be fine.

IsoaSFlus, 2021-04-24:
Any further progress? Is it possible to send the config packet via bluetooth directly?

NicoWeio, 2023-08-14:
@dos1, are you positive the panics were caused by your modifications? I'm asking because there have been some [fixes](https://github.com/torvalds/linux/commits/master/drivers/hid/hid-magicmouse.c) to the mainline kernel.

As mentioned elsewhere, [libratbag/piper](https://github.com/libratbag/piper/) ([libratbag/libratbag](https://github.com/libratbag/libratbag)) might be a good way to do this configuration in userspace.

ErikReider, 2024-09-23:
Sorry to bother, but are there any updates on this? :)