mwyborski/Linux-Magic-Trackpad-2-Driver #30: Not able to turn off acceleration

warptozero, 2018-12-08:
After installing the driver everything seems to work ok, except that acceleration is forced on. `libinput list-devices` gives `Accel profiles: none` instead of `Accel profiles: flat *adaptive`. So I'm unable to set it to flat.

warptozero, 2018-12-12:
Considering this patch has been merged into mainline, is this still the right place to report bugs?

robotrovsky, 2018-12-15:
@warptozero during merging to mainline we had to remove some offsets. The driver will need a libinput quirks file. As soon as 4.20 is released i will have a look at it. Anyways i had no problem setting or turning off the acceleration, neither with mtrack nor with libinput. Which distro are you using?

warptozero, 2018-12-17:
I'm using Arch Linux and I've compiled linux-mainline 4.20rc6 from the AUR.

Another problem I'm having is that upon resizing windows with the mouse (by holding down a keyboard button) the cursor jumps to the top left corner of the screen. Now this is usually caused by palm detection, however I'm unable to set `DisableWhileTyping` to false (in a xorg.conf file). As if the trackpad is not providing this option.

The cursor also feels a little laggier then should be, but this might just be the de/acceleration I'm feeling.

warptozero, 2018-12-17:
Well I've narrowed down the jumping cursor problem to setting the Coordinate Transformation Matrix with xinput:
```
xinput set-prop 'Apple Inc. Magic Trackpad 2' 'Coordinate Transformation Matrix' 1 0 0 0 1 0 0 0 -1
```
Using `-1 0 0 0 -1 0 0 0 1` also leads to the same behavior. The reason I'm doing this is that I like to use the trackpad upside down because I find it a lot more ergonomic (the default orientation gives me wrist pain). So far I haven't found an other way to switch orientation.

Would it be possible for the driver to provide an option for this? As it looks like this is a fairly common complaint and Apple provides a way to do this on macOS:

- `sudo defaults write com.apple.MultitouchSupport ForceAutoOrientation YES`
- Now place the Magic Trackpad in desired orientation on your desk and drop all 5 fingers on it (outstretched towards the edges not scrunched up).
- The coordinates of the trackpad automatically flip so all cursor and gesture motions match the new hand orientation.

warptozero, 2018-12-17:
If it's any help, here's the output of `xinput list-props`:
```
Device 'Apple Inc. Magic Trackpad 2':
	Device Enabled (143):	1
	Coordinate Transformation Matrix (145):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	libinput Tapping Enabled (279):	1
	libinput Tapping Enabled Default (280):	0
	libinput Tapping Drag Enabled (281):	1
	libinput Tapping Drag Enabled Default (282):	1
	libinput Tapping Drag Lock Enabled (283):	0
	libinput Tapping Drag Lock Enabled Default (284):	0
	libinput Tapping Button Mapping Enabled (285):	1, 0
	libinput Tapping Button Mapping Default (286):	1, 0
	libinput Natural Scrolling Enabled (287):	1
	libinput Natural Scrolling Enabled Default (288):	0
	libinput Scroll Methods Available (289):	1, 1, 0
	libinput Scroll Method Enabled (290):	1, 0, 0
	libinput Scroll Method Enabled Default (291):	1, 0, 0
	libinput Click Methods Available (292):	1, 1
	libinput Click Method Enabled (293):	1, 0
	libinput Click Method Enabled Default (294):	1, 0
	libinput Middle Emulation Enabled (295):	0
	libinput Middle Emulation Enabled Default (296):	0
	libinput Accel Speed (297):	0.000000
	libinput Accel Speed Default (298):	0.000000
	libinput Left Handed Enabled (299):	0
	libinput Left Handed Enabled Default (300):	0
	libinput Send Events Modes Available (264):	1, 0
	libinput Send Events Mode Enabled (265):	0, 0
	libinput Send Events Mode Enabled Default (266):	0, 0
	Device Node (267):	"/dev/input/event3"
	Device Product ID (268):	76, 613
	libinput Drag Lock Buttons (301):	<no items>
	libinput Horizontal Scroll Enabled (302):	1
```
No manual xinput changes.

And the only xorg.conf.d file that's active for pointers:
```
Section "InputClass"
    Identifier "libinput magic trackpad"
    MatchIsTouchpad "yes"
    MatchDevicePath "/dev/input/event*"
    MatchProduct "Magic Trackpad"
    Driver "libinput"

    # libinput options
    Option "AccelProfile" "flat"
    Option "NaturalScrolling" "True"
    Option "HorizontalScrolling" "True"
    Option "DisableWhileTyping" "False"
    Option "Tapping" "True"
EndSection
```
 If you like I can also paste the input parts of the Xorg.log.

warptozero, 2018-12-18:
@alanorth I haven't experienced a stuck cursor since I switched to 4.20, so maybe that could be because of manually compiling or dkms? I also experienced a once every few hours kernel lockup that disappeared with the switch.

To recap everything:

- Can't change acceleration profile
- Cursor jumps to top left corner of screen when Coordinate Transformation Matrix is changed from default (might be a libinput or xorg bug, but could be aleviated by providing an option to change trackpad orientation).
- Large amount of touch jumps detected and discarded when using touchpad normally (with only one finger touching): once or twice every few seconds and almost everytime when setting a finger down.