mwyborski/Linux-Magic-Trackpad-2-Driver #31: Raspberry Pi Kernel 4.19

Dygear, 2019-01-04:
@robotrovsky Thank you massively for your work on this.

Currently, the Raspberry Pi Foundation is looking at an upgrade of their mainline kernel to 4.19, [discussion here](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=224931&p=1411649#p1411649). Being that your patch landed in 4.20, is there any instructions that you could provide to getting this code working from within the Raspberry Pi 3 that I can run to get things like two finger scroll, and two finger tap, for right click?

Something along the lines of run this terminal command, do this terminal command and reboot, ect. I'm not really sure where north is on this one, as I've not really had to do something like this before.

robotrovsky, 2019-01-07:
@Dygear i think you can just check out commit 63187c13bb198609689460bbd1a99a4f622b5d83
This is the last commit before the pressure tweaks for the mainline release have been removed. Just check it out and go to ./linux/drivers/hid and do a make. I haven't tried it with 4.19, but if it doesn't compile i would help you. You can also check out the current official release, but then you would need to make a libinput quirks file, because the pressures wouldn't be recognized. So i think this would be the easiest way to try it.

```
cd linux/drivers/hid
make
sudo rmmod hid_magicmouse
sudo insmod ./hid-magicmouse.ko
tail -f ~/.local/share/xorg/Xorg.0.log
```

robotrovsky, 2019-01-07:
I also added info about the necessary quirks file for the current release to the README, so you could try it as well with the current release and add the libinput quirks file

closed 2019-04-11 by robotrovsky