mwyborski/Linux-Magic-Trackpad-2-Driver #32: Needs too much presure (Ubuntu 18.04)

tasos14, 2019-01-12:
I'm on Ubuntu 18.04 with 4.20 kernel, on wireless mode scrolling needs a lot of pressure. When connected it works fine.
I have added the quirks file to `/usr/share/libinput/.quirks` as suggested by this comment https://github.com/robotrovsky/Linux-Magic-Trackpad-2-Driver/issues/24#issuecomment-434697180
But the problem still persists.

dos1, 2019-01-12:
Use `libinput quirks list /dev/input/eventX` to verify whether the quirks file is really applied.

For the reference, this is my `/etc/libinput/local-overrides.quirks` that works for me in both modes:
```
[Touchpad pressure override]
MatchName=*Magic Trackpad 2
AttrPressureRange=2:0
```

dos1, 2019-01-12:
You may also try `libinput list-quirks`, I think that command got renamed recently.

closed 2019-04-11 by robotrovsky

whot, 2019-04-30:
bug for libinput was filed here: https://gitlab.freedesktop.org/libinput/libinput/issues/280