mwyborski/Linux-Magic-Trackpad-2-Driver #33: Magic Mouse 2 - Scroll not working

romanAbrahamovic, 2019-01-28:
I finish instalation by instructions in readme file. 

I still cannot scroll on my MM2. Is there a something which I missing? 

```
kernel: 
4.18.5-041805-generic
```

```
+ dkms_name=hid-magicmouse-dkms
+ dkms_version=4.18+magictrackpad2
+++ dirname ./install.sh
++ cd .
++ pwd
+ DIR=/***/***/Linux-Magic-Trackpad-2-Driver/scripts
+ dkms status -m hid-magicmouse-dkms -v 4.18+magictrackpad2
+ egrep '(added|built|installed)'
+ dkms status -m hid-magicmouse-dkms -v 4.18+magictrackpad2
+ egrep '(built|installed)'
+ dkms status -m hid-magicmouse-dkms -v 4.18+magictrackpad2
+ egrep '(installed)'
```

If I use kernel v.4.20 or 4.15 still have a same problem with scroll. Thank you for advice.

Liemaeu, 2019-03-13:
I have the same problem (Ubuntu 18.04.2, 4.18.0-16-generic)

robotrovsky, 2019-03-13:
You mean the Magic Trackpad 2, not the Magic Mouse 2, right?
You can just checkout the version below and you dont need the quirks file, everything should work:
https://github.com/robotrovsky/Linux-Magic-Trackpad-2-Driver/tree/63187c13bb198609689460bbd1a99a4f622b5d83

romanAbrahamovic, 2019-03-13:
@robotrovsky Magic Mouse 2, not Magic Trackpad 2. Support this drive magic mouse 2?

robotrovsky, 2019-03-18:
@romanAbrahamovic no this driver doesn't support the Magic Mouse 2. Only Magic Mouse I and Magic Trackpad I + II

closed 2019-03-18 by robotrovsky

kubaplas, 2019-04-24:
I can confirm that the driver works with Magic Mouse 2, but it needs to be manualy loaded - 
see https://forum.manjaro.org/t/apple-magic-mouse-2-not-working-kernel-4-20/72236/5

Liemaeu, 2019-06-17:
@stubbulon5 Me for example!

DaBlitzStein, 2019-08-03:
Scroll not working on 5.0.23 ubuntu 18.04 .

stubbulon5, 2019-08-23:
@robotrovsky , is seems Magic Trackpad 2 scrolling works but not Magic Mouse 2 scrolling. How feasible would a fix for this be? 

Much respect for the work you've done so far.

gchehab, 2019-10-14:
I am using this fork patched to work with magic mouse 2: [https://github.com/rohitpid/Linux-Magic-Trackpad-2-Driver](https://github.com/rohitpid/Linux-Magic-Trackpad-2-Driver).
The (virtual) wheel using single finger swipe works ok using libinput driver option.
It would be nice to have it merged with the main version -- I guess I could do a merge request, should it be acceptable.

txtad, 2019-12-07:
> It would be nice to have it merged with the main version -- I guess I could do a merge request, should it be acceptable.

That would be wonderful. I've converted a Mac mini to an Intel Nuc running Ubuntu, but I've kept the mini's keyboard and mouse. I was happily surprised that they are working at the login screen, but then quickly disappointed that scrolling with the Magic Mouse 2 wasn't working. I was honestly surprised that it didn't "just work".

leftdevel, 2020-05-29:
Sorry for posting on an old thread. I got the magic mouse working even with the scroll feature, but scrolling is quite slow, just to scroll this thread it takes me 7 swipes. Do you know how I can increase the scroll speed?

Thank you.

imanm93, 2020-07-26:
@leftdevel could you please describe how you got the scrolling to work on the magic mouse 2?

Also will there be any updates to support this for ubuntu 20.04 - scrolling doesn't seem to work on this version (kernel v5.4.0-42-generic

Thanks in advance!

maomorales, 2020-07-30:
@imanm93 follow the instructions from rohitpid branch's readme. Make/compile and then run:

sudo rmmod hid-magicmouse
sudo insmod ./hid-magicmouse.ko middle_click_3finger=1

That worked form in 20.04, enjoy the scrolling! :D

akozyreva, 2020-10-27:
@maomorales for me your decision works, but every time, when mouse is disconnected with bluetooth or computer sleeps for a while, scrolling doesn't work again and I need every time run these commands
```
sudo rmmod hid-magicmouse
sudo insmod ./hid-magicmouse.ko middle_click_3finger=1
```
Does anybody have the same execution? How to fix it?

maomorales, 2020-10-27:
@akozyreva I ended up changing the Mouse. I couldn't solve it and the time I was spending there was not a smart use of my time 😄