mwyborski/Linux-Magic-Trackpad-2-Driver #34: How to get the battery state?

schmunk42, 2019-02-08:
My problem is that I currently need to have my touchpad almost always connected via USB to have it working. I suppose this is due a low battery state, but I am not sure since I can't see the loading percentage at the moment.

I found only https://github.com/robotrovsky/Linux-Magic-Trackpad-2-Driver/issues/19 when looking for "battery", do you have any advice how to get the actual battery level?

dos1, 2019-02-21:
https://github.com/robotrovsky/Linux-Magic-Trackpad-2-Driver/issues/19#issuecomment-409420052 describes it pretty well. You have to do it manually, as it's not implemented in the driver yet.

aleksfadini, 2021-03-11:
I also would like to have any way to know the battery state of my magic trackpad 2 on linux. Even if it means being able to check only when it is connected to a wire, that would still be a big improvement. As things stand now, we have no way to tell really.

alexanderadam, 2023-08-10:
In Gnome you can check the power state of devices under **Settings** > **Power** > **Devices** without having to install or configure anything:

[image]

mwyborski, 2023-08-10:
@alexanderadam is right, that modifications should be made in the official linux kernel if necessary. Thank you

alexanderadam, 2023-08-10:
@mwyborski I think that you can close this issue here, too, right?

NicoWeio, 2023-08-14:
For future reference, battery level reporting (USB only) was added in https://github.com/torvalds/linux/commit/0b91b4e4dae63cd43871fc2012370b86ee588f91. 
It also works for me on KDE Plasma, so this issue could indeed be closed.

closed 2023-08-15 by mwyborski