mwyborski/Linux-Magic-Trackpad-2-Driver #35: How to disable USB and only use Bluetooth?

schmunk42, 2019-02-21:
I noticed that since I have connected my TrackPad via USB mainly due to loading it (#34) it also uses USB as the "main control channel". Since then I am experiencing kernel panics from time to time - I don't know if this is directly related, but I'd like to know if there's a way to disable the USB control connection and **only** load via USB?

robotrovsky, 2019-02-21:
Hi! I dont know of a good way to achieve this. I think as soon as the Trackpad finds a USB-connection it will switch off Bluetooth. But honestly i dont believe the kernel panics come from the Trackpad. Do you use 4.20? And which Distribution? I had problems with 4.20 and a freezing system as well, but without the Trackpad being connected. Maybe you can connect the Trackpad to a USB-phone-charger and see if it works better for you. That said, the Trackpad seems to work a little more responsive over USB than over Bluetooth not only on Linux but on OSX as well.
Good luck!

closed 2019-02-21 by robotrovsky

schmunk42, 2019-02-21:
> Do you use 4.20? 

yes

> And which Distribution?

Antergos/Arch

```
Linux xps-15 4.20.8-arch1-1-ARCH #1 SMP PREEMPT Wed Feb 13 00:06:58 UTC 2019 x86_64 GNU/Linux
```

robotrovsky, 2019-02-21:
I also have a xps-15 and kernel panics with 4.20. At least last time i tried it like 4.20.5 or something. But i haven't had the Trackpad connected, so if you ask me, i think it is because of 4.20.

schmunk42, 2019-02-21:
Thank you for the feedback, my issues definitely arose since 4.20.

robotrovsky, 2019-02-21:
You are welcome ;-)