mwyborski/Linux-Magic-Trackpad-2-Driver #36: Ubuntu 18.04.2 no scrolling

Liemaeu, 2019-03-13:
My System:
Ubuntu 18.04.2 (amd64)
Kernel: 4.18.0-16-generic

I can't scroll with my Magic Mouse 2. 

There was no libinput directory in /etc, so I created it (sudo mkdir /etc/libinput) and placed the local-overrides.quirks file in it. But it still doesn't work.

robotrovsky, 2019-03-13:
Have you compiled the kernel module yourself? It is not included in 4.18 it is included from 4.20 onwards.
You can just checkout the version below and you dont need the quirks file, everything should work:
https://github.com/robotrovsky/Linux-Magic-Trackpad-2-Driver/tree/63187c13bb198609689460bbd1a99a4f622b5d83

tegk, 2019-04-10:
Same here with Linux 5.0.7-050007-generic on Ubuntu 18.10. Mouse works with right and left click but no scrolling. Any tips to get it working?

dos1, 2019-04-10:
At first glance I'd say that it's probably unrelated to the driver, but to configuration of input driver in your display servers/desktop environments.

robotrovsky, 2019-04-11:
I recommend to wait a few days and upgrade to Ubuntu 19.04, there everything works out of the box.

closed 2019-04-11 by robotrovsky