mwyborski/Linux-Magic-Trackpad-2-Driver #38: Magic Mouse Scrolling not working in Ubuntu 19.04

ddalida, 2019-05-02:
I had this module working in 18.04 with the 4.20... kernel without issues. I know it has been implemented into the release but the scrolling isn't working. I've tried rmmod and insmod under the 5.0.0-13 kernel with 19.04 but no joy. This is with the Magic Mouse 2 and not the trackpad.

Am I missing something? Any info that could be relevant I can provide?

ddalida, 2019-05-02:
This is what gets loaded at startup (lsmod | grep hid):

hid_apple              16384  0
hidp                   28672  2
hid_generic            16384  0
bluetooth             557056  40 btrtl,hidp,btintel,btbcm,bnep,btusb,rfcomm
usbhid                 53248  0
mac_hid                16384  0
hid_magicmouse         20480  0
hid                   126976  5 hidp,usbhid,hid_apple,hid_generic,hid_magicmouse

stubbulon5, 2019-06-17:
(--Moving from a closed issue--)

@romanAbrahamovic how much work would supporting Magic Mouse 2 take? (Enabling scrolling, that is)

I recon there would be many Linux users happy to buy you a coffee if you got this working...

djhoese, 2019-09-05:
I found that there is a branch in @rohitpid's fork that is at least a start on getting MM2 to work, but I haven't tested it: https://github.com/rohitpid/Linux-Magic-Trackpad-2-Driver/tree/magicmouse2

stubbulon5, 2019-09-27:
> I found that there is a branch in @rohitpid's fork that is at least a start on getting MM2 to work, but I haven't tested it: https://github.com/rohitpid/Linux-Magic-Trackpad-2-Driver/tree/magicmouse2

I tested it today, and it worked! thanks @rohitpid

jeffywu, 2020-01-02:
Hi, wondering if anyone can help with debugging an issue on this magicmouse2 driver. The mouse connects over bluetooth successfully every time but the scroll wheel does not work until I reload the driver using:
```
sudo rmmod hid_magicmouse
sudo insmod /lib/modules/5.3.0-7625-generic/updates/dkms/hid-magicmouse.ko
```

These are the logs in `~/.local/share/xorg/Xorg.1.log`:
```
[    30.156] (II) config/udev: Adding input device Home Mouse (/dev/input/mouse1)
[    30.156] (II) No input driver specified, ignoring this device.
[    30.156] (II) This device may have been added with another device file.
[    30.212] (II) config/udev: Adding input device Home Mouse (/dev/input/event14)
[    30.212] (**) Home Mouse: Applying InputClass "libinput pointer catchall"
[    30.212] (II) Using input driver 'libinput' for 'Home Mouse'
[    30.213] (II) systemd-logind: got fd for /dev/input/event14 13:78 fd 60 paused 0
[    30.213] (**) Home Mouse: always reports core events
[    30.213] (**) Option "Device" "/dev/input/event14"
[    30.213] (**) Option "_source" "server/udev"
[    30.217] (II) event14 - Home Mouse: is tagged by udev as: Mouse
[    30.217] (EE) event14 - Home Mouse: libinput bug: kernel fuzz of 4 but LIBINPUT_FUZZ_35 is missing
[    30.217] (EE) event14 - Home Mouse: libinput bug: kernel fuzz of 4 but LIBINPUT_FUZZ_36 is missing
[    30.217] (II) event14 - Home Mouse: device is a pointer
[    30.218] (II) event14 - Home Mouse: device removed
[    30.218] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:256/0005:004C:0269.0001/input/input23/event14"
[    30.218] (II) XINPUT: Adding extended input device "Home Mouse" (type: MOUSE, id 12)
[    30.220] (**) Option "AccelerationScheme" "none"
[    30.221] (**) Home Mouse: (accel) selected scheme none/0
[    30.221] (**) Home Mouse: (accel) acceleration factor: 2.000
[    30.221] (**) Home Mouse: (accel) acceleration threshold: 4
[    30.224] (II) event14 - Home Mouse: is tagged by udev as: Mouse
[    30.224] (EE) event14 - Home Mouse: libinput bug: kernel fuzz of 4 but LIBINPUT_FUZZ_35 is missing
[    30.224] (EE) event14 - Home Mouse: libinput bug: kernel fuzz of 4 but LIBINPUT_FUZZ_36 is missing
[    30.225] (II) event14 - Home Mouse: device is a pointer
[    61.796] (II) config/udev: removing device Home Mouse
[    61.796] (**) Option "fd" "60"
[    61.796] (II) event14 - Home Mouse: device removed
[    61.803] (II) UnloadModule: "libinput"
[    61.803] (II) systemd-logind: releasing fd for 13:78
[    61.830] (II) config/udev: Adding input device Home Mouse (/dev/input/mouse1)
[    61.831] (II) No input driver specified, ignoring this device.
[    61.831] (II) This device may have been added with another device file.
[    61.875] (II) config/udev: Adding input device Home Mouse (/dev/input/event14)
[    61.876] (**) Home Mouse: Applying InputClass "libinput pointer catchall"
[    61.876] (II) Using input driver 'libinput' for 'Home Mouse'
[    61.889] (EE) systemd-logind: failed to take device /dev/input/event14: No such device
[    61.889] (**) Home Mouse: always reports core events
[    61.889] (**) Option "Device" "/dev/input/event14"
[    61.889] (**) Option "_source" "server/udev"
[    61.889] (EE) xf86OpenSerial: Cannot open device /dev/input/event14
	Permission denied.
[    61.889] (II) event14: opening input device '/dev/input/event14' failed (Permission denied).
[    61.889] (II) event14 - failed to create input device '/dev/input/event14'.
[    61.889] (EE) libinput: Home Mouse: Failed to create a device for /dev/input/event14
[    61.889] (EE) PreInit returned 2 for "Home Mouse"
[    61.889] (II) UnloadModule: "libinput"
[    61.891] (II) config/udev: Adding input device Home Mouse (/dev/input/event15)
[    61.891] (II) No input driver specified, ignoring this device.
[    61.891] (II) This device may have been added with another device file.
[    61.984] (II) config/udev: Adding input device Home Mouse (/dev/input/mouse1)
[    61.984] (II) No input driver specified, ignoring this device.
[    61.984] (II) This device may have been added with another device file.
[    62.060] (II) config/udev: Adding input device Home Mouse (/dev/input/event14)
[    62.060] (**) Home Mouse: Applying InputClass "libinput pointer catchall"
[    62.060] (II) Using input driver 'libinput' for 'Home Mouse'
[    62.066] (II) systemd-logind: got fd for /dev/input/event14 13:78 fd 68 paused 0
[    62.066] (**) Home Mouse: always reports core events
[    62.066] (**) Option "Device" "/dev/input/event14"
[    62.066] (**) Option "_source" "server/udev"
[    62.069] (II) event14 - Home Mouse: is tagged by udev as: Mouse
[    62.069] (EE) event14 - Home Mouse: libinput bug: kernel fuzz of 4 but LIBINPUT_FUZZ_35 is missing
[    62.069] (EE) event14 - Home Mouse: libinput bug: kernel fuzz of 4 but LIBINPUT_FUZZ_36 is missing
[    62.070] (II) event14 - Home Mouse: device is a pointer
[    62.070] (II) event14 - Home Mouse: device removed
[    62.070] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:256/0005:004C:0269.0001/input/input26/event14"
[    62.070] (II) XINPUT: Adding extended input device "Home Mouse" (type: MOUSE, id 12)
[    62.072] (**) Option "AccelerationScheme" "none"
[    62.074] (**) Home Mouse: (accel) selected scheme none/0
[    62.074] (**) Home Mouse: (accel) acceleration factor: 2.000
[    62.074] (**) Home Mouse: (accel) acceleration threshold: 4
[    62.077] (II) event14 - Home Mouse: is tagged by udev as: Mouse
[    62.077] (EE) event14 - Home Mouse: libinput bug: kernel fuzz of 4 but LIBINPUT_FUZZ_35 is missing
[    62.077] (EE) event14 - Home Mouse: libinput bug: kernel fuzz of 4 but LIBINPUT_FUZZ_36 is missing
[    62.077] (II) event14 - Home Mouse: device is a pointer
```

djhoese, 2020-01-02:
I had the same issue and my mouse would randomly disconnect. I ended up getting rid of my magic mouse and switched to a logitech bluetooth mouse after I couldn't find a solution to the scrolling issue. You may have luck with the "unstable" branch mentioned above.

RicardoEPRodrigues, 2020-02-03:
Hello, I have been scrolling through several of these issues and found some solutions, I decided to pick some of them up and compiled them. Can you please try my solution out?

https://github.com/RicardoEPRodrigues/Linux-Magic-Trackpad-2-Driver

jeffywu, 2020-02-05:
@RicardoEPRodrigues, thanks! I tested out your instructions and they worked well. A couple minor suggestions. This udev configuration works well for me, in your version the second line is missing a comma and a backslash I believe.

```
SUBSYSTEMS=="input", \
    ATTRS{name}=="XXX", \
    ACTION=="add", \
    SYMLINK+="input/magicmouse", \
    RUN+="/opt/magic-mouse-fix/magic-mouse-2-add.sh"
```

I was able to view the properties on the mouse via dbus:
`dbus-send --system --print-reply --type=method_call --dest=org.bluez / org.freedesktop.DBus.ObjectManager.GetManagedObjects`

RicardoEPRodrigues, 2020-02-05:
Hello, thank you for the input! I updated the `udev` rule. I hope it now works correctly:

```
SUBSYSTEMS=="usb", \
    ATTRS{idProduct}=="e300", \
    ACTION=="add", \
    SYMLINK+="input/magicmouse", \
    RUN+="/opt/magic-mouse-fix/magic-mouse-2-add.sh"
```

Correct me if I'm wrong, but I think this rule goes like: "if an usb device with the product id of `e300` is detected (the ID is the one from the Magic Mouse) then load an input driver for that device".

I read more info on the device based on this command `udevadm info -a -p $(udevadm info -q path -n /dev/input/eventXX)`. Change the `/dev/input/eventXX` with the correct device in your OS.

RicardoEPRodrigues, 2020-02-06:
Hello again, sorry for the SPAM.

My solution doesn't work as is. The way I was trying to do things was to read an id from `usb1/X-X` which is specific to the PC and not the mouse. 

I trying to find a solution. If we can find a Magic Mouse 2 ID in the `udev` info then we can change the rule and the solution should work for everyone.

If not, then I'm trying to find an ID in the Bluetooth info, with that we can have a script running every time an input device is plugged in, but that script will only load the drive if that device is a Magic Mouse 2.

You can help me by posting a file with the output of this command `udevadm info -a -p $(udevadm info -q path -n /dev/input/eventXX)`, where you need to change the `/dev/input/eventXX` with the `XX` corresponding with the Magic Mouse 2.

RicardoEPRodrigues, 2020-02-06:
Ok, I figured it out.

I rewrote the rule to catch all the Magic Mouse 2. FINALLY. Check this out:

```
SUBSYSTEM=="input", \
    KERNEL=="mouse*", \
    KERNELS=="0005:004C:0269*", \
    ACTION=="add", \
    SYMLINK+="input/magicmouse-%k", \
    RUN+="/opt/magic-mouse-fix/magic-mouse-2-add.sh"
```

Remember to try my `./install` in my repo. Any input is good input.

jeffywu, 2020-02-10:
The second rule works for me. Thanks for your effort!

ralphsc, 2020-06-11:
This isn't working on Elementary OS 5.1.5. Here's the steps I'm taking. Move the original kernel driver hid-magicmouse.ko from my 5.3.0-59-generic kernel and running "depmod -a" then reboot. After the system comes back up I'm following the install instructions (git clone and then running the install.sh as root) and reboot. When the system is back up and I click the mouse it is seen and paired but the pointer doesn't move, the track-pad still works though. Here's the output from my xorg.log:

[   546.767] (II) config/udev: Adding input device (unnamed) (/dev/input/mouse0)
[   546.767] (II) No input driver specified, ignoring this device.
[   546.767] (II) This device may have been added with another device file.
[   609.625] (EE) event16 - bcm5974: kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.10.4/touchpad_jumping_cursor.html for details
[   692.303] (II) config/udev: Adding input device Magic Mouse 2 (/dev/input/event7)
[   692.303] (**) Magic Mouse 2: Applying InputClass "libinput pointer catchall"
[   692.303] (II) Using input driver 'libinput' for 'Magic Mouse 2'
[   692.303] (**) Magic Mouse 2: always reports core events
[   692.303] (**) Option "Device" "/dev/input/event7"
[   692.303] (**) Option "_source" "server/udev"
[   692.322] (EE) xf86OpenSerial: Cannot open device /dev/input/event7
        No such device.
[   692.322] (II) event7: opening input device '/dev/input/event7' failed (No such device).
[   692.322] (II) event7  - failed to create input device '/dev/input/event7'.
[   692.322] (EE) libinput: Magic Mouse 2: Failed to create a device for /dev/input/event7
[   692.322] (EE) PreInit returned 2 for "Magic Mouse 2"
[   692.322] (II) UnloadModule: "libinput"

Please let me know if you need more info from me.
Thanks,
Ralph

RicardoEPRodrigues, 2020-06-12:
@ralphsc did you try my fork?

https://github.com/RicardoEPRodrigues/Linux-Magic-Trackpad-2-Driver

I have a pending pull-request (https://github.com/rohitpid/Linux-Magic-Trackpad-2-Driver/pull/1) where some discussion has been going on. It might help.

ralphsc, 2020-06-12:
> I have a pending pull-request (rohitpid#1

Hi Ricardo,
Yes I was using your fork for this.

I'll take a look at
https://github.com/rohitpid/Linux-Magic-Trackpad-2-Driver/pull/1  and let
you know how it goes.

ralphsc, 2020-06-12:
Ricardo,
I tried the suggestions in the link you sent but it's still not working.
This time I installed by running install-fix.sh, the mouse now moves but
without scrolling. I also removed / moved the original kernel
hid-magicmouse.ko driver before starting. Here's the xorg log when the
mouse is loaded:

[   805.929] (II) config/udev: Adding input device Magic Mouse 2
(/dev/input/mouse0)
[   805.929] (II) No input driver specified, ignoring this device.
[   805.929] (II) This device may have been added with another device file.
[   806.011] (II) config/udev: Adding input device Magic Mouse 2
(/dev/input/event7)
[   806.011] (**) Magic Mouse 2: Applying InputClass "libinput pointer
catchall"
[   806.011] (II) Using input driver 'libinput' for 'Magic Mouse 2'
[   806.011] (**) Magic Mouse 2: always reports core events
[   806.012] (**) Option "Device" "/dev/input/event7"
[   806.012] (**) Option "_source" "server/udev"
[   806.012] (II) event7  - Magic Mouse 2: is tagged by udev as: Mouse
[   806.012] (II) event7  - Magic Mouse 2: device is a pointer
[   806.012] (II) event7  - Magic Mouse 2: device removed
[   806.067] (**) Option "config_info"
"udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.3/1-8.3:1.0/bluetooth/hci0/hci0:12/0005:004C:0269.000A/input/input225/event7"
[   806.067] (II) XINPUT: Adding extended input device "Magic Mouse 2"
(type: MOUSE, id 13)
[   806.068] (**) Option "AccelerationScheme" "none"
[   806.069] (**) Magic Mouse 2: (accel) selected scheme none/0
[   806.069] (**) Magic Mouse 2: (accel) acceleration factor: 2.000
[   806.069] (**) Magic Mouse 2: (accel) acceleration threshold: 4
[   806.070] (II) event7  - Magic Mouse 2: is tagged by udev as: Mouse
[   806.070] (II) event7  - Magic Mouse 2: device is a pointer

ralphsc, 2020-06-12:
And just so you can see what happens with the udev rule in place here's the
xorg log output when the mouse is connected:

[  1845.138] (II) config/udev: Adding input device Magic Mouse 2
(/dev/input/event7)
[  1845.139] (**) Magic Mouse 2: Applying InputClass "libinput pointer
catchall"
[  1845.139] (II) Using input driver 'libinput' for 'Magic Mouse 2'
[  1845.139] (**) Magic Mouse 2: always reports core events
[  1845.139] (**) Option "Device" "/dev/input/event7"
[  1845.139] (**) Option "_source" "server/udev"
[  1845.155] (EE) xf86OpenSerial: Cannot open device /dev/input/event7
No such device.
[  1845.155] (II) event7: opening input device '/dev/input/event7' failed
(No such device).
[  1845.155] (II) event7  - failed to create input device
'/dev/input/event7'.
[  1845.155] (EE) libinput: Magic Mouse 2: Failed to create a device for
/dev/input/event7
[  1845.155] (EE) PreInit returned 2 for "Magic Mouse 2"
[  1845.155] (II) UnloadModule: "libinput"
[  1849.370] (II) config/udev: Adding input device (unnamed)
(/dev/input/mouse0)
[  1849.370] (II) No input driver specified, ignoring this device.
[  1849.370] (II) This device may have been added with another device file.

RicardoEPRodrigues, 2020-06-14:
@ralphsc Unfortunately I don't have a lot of time to try to debug this issue, but here are my thoughts.

The Magic Mouse 2 creates 2 devices, one `/dev/input/eventX` (the touchpad) and a `/dev/input/mouseX` (the mouse and buttons). `libinput` picks up any `mouseX` event and loads a driver for it, that is why by default you have a mouse with no scroll.

What my repo does is to load the driver for the `eventX` that knows how to deal with the touchpad.

I believe that for some reason, either the rule isn't being called and that is why you have a mouse moving and no scroll, or the udev rules are removing the `mouseX`, but still loading the driver, which results in the touchpad working, but not the mouse.

If you have some time, please read through [0xABAD repo](https://github.com/0xABAD/magic-mouse-2) where the process to get the driver loaded is described. My repo is inspired and improves on his work.

ralphsc, 2020-06-15:
Hey Ricardo,
No worries man. Thanks for the info I'll read through it and if I do figure
it out I'll share what works with this thread.

Thanks again for your time.

Sincerely,
Ralph

ralphsc, 2020-06-18:
On Elementary OS I had to add a DRIVER section to get it to work, like so:

SUBSYSTEM=="input", \
    KERNEL=="mouse*", \
    KERNELS=="0005:004C:0269.*"
    SUBSYSTEMS=="hid", \
    DRIVERS=="hid-generic", \
    ACTION=="add", \
    SYMLINK+="input/magicmouse-%k", \
    RUN+="/opt/magic-mouse-fix/magic-mouse-2-add.sh"

RicardoEPRodrigues, 2020-06-18:
@ralphsc Great to know you got it working!

I tried to use your `udev` rule and it didn't work for me. I don't have any parent with `DRIVERS=="hid-generic"`. I made some changes, can you try to run this version and check if it works?

```
SUBSYSTEM=="input", \
    KERNEL=="mouse*", \
    DRIVER=="", \
    SUBSYSTEMS=="hid", \
    KERNELS=="0005:004C:0269*", \
    DRIVERS=="magicmouse", \
    ACTION=="add", \
    SYMLINK+="input/magicmouse-%k", \
    RUN+="/opt/magic-mouse-fix/magic-mouse-2-add.sh"
```

Thank you!

ralphsc, 2020-06-18:
I tried it and it didn't work for me. If I change the udev rule, disconnect
the mouse, unload the driver and click the mouse it pair's with BT and is
connected but no mouse movement. When I change the rule back and disconnect
the mouse, unload the driver and click the mouse it pair's with BT and it
all works.

This is on:
elementary OS 5.1.5 Hera
5.3.0-59-generic
On a MAC Book Pro 2013 era.

RicardoEPRodrigues, 2020-06-18:
Can you show me the output of `udevadm info -a -p $(udevadm info -q path -n /dev/input/mouseX)` where `mouseX` is the connected magic mouse?

I edited my previous answer to have ` DRIVERS=="hid-generic|magicmouse"`, you used that one right?

Can you try the following option (I removed the `DRIVER==""`) :

```
SUBSYSTEM=="input", \
    KERNEL=="mouse*", \
    SUBSYSTEMS=="hid", \
    KERNELS=="0005:004C:0269*", \
    DRIVERS=="hid-generic|magicmouse", \
    ACTION=="add", \
    SYMLINK+="input/magicmouse-%k", \
    RUN+="/opt/magic-mouse-fix/magic-mouse-2-add.sh"
```

Also, what is the version of you `udev`? You can check it with `apt show udev`.

ralphsc, 2020-06-18:
udev:
Version: 237-3ubuntu10.41
xinput:
Version: 1.6.2-1build1

ralphsc, 2020-06-18:
It didn't work with that rule either. It seems that if I have "magicmouse" as a driver it fails to function.
Here's the hid drivers I have loaded when the mouse is working:

lsmod | grep hid
hid_magicmouse         20480  0
hidp                   28672  2
bluetooth             573440  42 btrtl,hidp,btintel,btbcm,bnep,btusb,rfcomm
mac_hid                16384  0
hid_apple              16384  0
hid_generic            16384  0
usbhid                 53248  0
hid                   126976  5 hidp,usbhid,hid_apple,hid_generic,hid_magicmouse

RicardoEPRodrigues, 2020-06-19:
Can you show me the output of `udevadm info -a -p $(udevadm info -q path -n /dev/input/mouseX)` where `mouseX` is the connected magic mouse?

ralphsc, 2020-06-19:
Sure thing. FYI, it does show DRIVERS=="magicmouse" but it doesn't work in
my udev rule.

udevadm info -a -p $(udevadm info -q path -n /dev/input/event6)

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device
'/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.3/1-8.3:1.0/bluetooth/hci0/hci0:11/0005:004C:0269.0017/input/input143/event6':
    KERNEL=="event6"
    SUBSYSTEM=="input"
    DRIVER==""

  looking at parent device
'/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.3/1-8.3:1.0/bluetooth/hci0/hci0:11/0005:004C:0269.0017/input/input143':
    KERNELS=="input143"
    SUBSYSTEMS=="input"
    DRIVERS==""
    ATTRS{name}=="Magic Mouse 2"
    ATTRS{phys}=="3c:15:c2:da:0a:27"
    ATTRS{properties}=="0"
    ATTRS{uniq}=="ac:bc:32:e6:54:80"

  looking at parent device
'/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.3/1-8.3:1.0/bluetooth/hci0/hci0:11/0005:004C:0269.0017':
    KERNELS=="0005:004C:0269.0017"
    SUBSYSTEMS=="hid"
    DRIVERS=="magicmouse"
    ATTRS{country}=="00"

  looking at parent device
'/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.3/1-8.3:1.0/bluetooth/hci0/hci0:11':
    KERNELS=="hci0:11"
    SUBSYSTEMS=="bluetooth"
    DRIVERS==""

  looking at parent device
'/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.3/1-8.3:1.0/bluetooth/hci0':
    KERNELS=="hci0"
    SUBSYSTEMS=="bluetooth"
    DRIVERS==""

  looking at parent device
'/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.3/1-8.3:1.0':
    KERNELS=="1-8.3:1.0"
    SUBSYSTEMS=="usb"
    DRIVERS=="btusb"
    ATTRS{authorized}=="1"
    ATTRS{bAlternateSetting}==" 0"
    ATTRS{bInterfaceClass}=="ff"
    ATTRS{bInterfaceNumber}=="00"
    ATTRS{bInterfaceProtocol}=="01"
    ATTRS{bInterfaceSubClass}=="01"
    ATTRS{bNumEndpoints}=="03"
    ATTRS{supports_autosuspend}=="1"

  looking at parent device
'/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.3':
    KERNELS=="1-8.3"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="ff"
    ATTRS{bDeviceProtocol}=="01"
    ATTRS{bDeviceSubClass}=="01"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{bMaxPower}=="0mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 4"
    ATTRS{bcdDevice}=="0137"
    ATTRS{bmAttributes}=="e0"
    ATTRS{busnum}=="1"
    ATTRS{configuration}==""
    ATTRS{devnum}=="6"
    ATTRS{devpath}=="8.3"
    ATTRS{idProduct}=="8289"
    ATTRS{idVendor}=="05ac"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Apple Inc."
    ATTRS{maxchild}=="0"
    ATTRS{product}=="Bluetooth USB Host Controller"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="fixed"
    ATTRS{rx_lanes}=="1"
    ATTRS{speed}=="12"
    ATTRS{tx_lanes}=="1"
    ATTRS{urbnum}=="127628"
    ATTRS{version}==" 2.00"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1/1-8':
    KERNELS=="1-8"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="09"
    ATTRS{bDeviceProtocol}=="00"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPacketSize0}=="8"
    ATTRS{bMaxPower}=="94mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 1"
    ATTRS{bcdDevice}=="0100"
    ATTRS{bmAttributes}=="e0"
    ATTRS{busnum}=="1"
    ATTRS{configuration}==""
    ATTRS{devnum}=="2"
    ATTRS{devpath}=="8"
    ATTRS{idProduct}=="4500"
    ATTRS{idVendor}=="0a5c"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Apple Inc."
    ATTRS{maxchild}=="3"
    ATTRS{product}=="BRCM20702 Hub"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="unknown"
    ATTRS{rx_lanes}=="1"
    ATTRS{speed}=="12"
    ATTRS{tx_lanes}=="1"
    ATTRS{urbnum}=="4309"
    ATTRS{version}==" 2.00"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1':
    KERNELS=="usb1"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{authorized_default}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="09"
    ATTRS{bDeviceProtocol}=="01"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{bMaxPower}=="0mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 1"
    ATTRS{bcdDevice}=="0503"
    ATTRS{bmAttributes}=="e0"
    ATTRS{busnum}=="1"
    ATTRS{configuration}==""
    ATTRS{devnum}=="1"
    ATTRS{devpath}=="0"
    ATTRS{idProduct}=="0002"
    ATTRS{idVendor}=="1d6b"
    ATTRS{interface_authorized_default}=="1"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Linux 5.3.0-59-generic xhci-hcd"
    ATTRS{maxchild}=="14"
    ATTRS{product}=="xHCI Host Controller"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="unknown"
    ATTRS{rx_lanes}=="1"
    ATTRS{serial}=="0000:00:14.0"
    ATTRS{speed}=="480"
    ATTRS{tx_lanes}=="1"
    ATTRS{urbnum}=="2017"
    ATTRS{version}==" 2.00"

  looking at parent device '/devices/pci0000:00/0000:00:14.0':
    KERNELS=="0000:00:14.0"
    SUBSYSTEMS=="pci"
    DRIVERS=="xhci_hcd"
    ATTRS{ari_enabled}=="0"
    ATTRS{broken_parity_status}=="0"
    ATTRS{class}=="0x0c0330"
    ATTRS{consistent_dma_mask_bits}=="64"
    ATTRS{d3cold_allowed}=="1"
    ATTRS{device}=="0x8c31"
    ATTRS{dma_mask_bits}=="64"
    ATTRS{driver_override}=="(null)"
    ATTRS{enable}=="1"
    ATTRS{irq}=="37"
    ATTRS{local_cpulist}=="0-7"
    ATTRS{local_cpus}=="ff"
    ATTRS{msi_bus}=="1"
    ATTRS{numa_node}=="-1"
    ATTRS{revision}=="0x05"
    ATTRS{subsystem_device}=="0x7270"
    ATTRS{subsystem_vendor}=="0x8086"
    ATTRS{vendor}=="0x8086"

  looking at parent device '/devices/pci0000:00':
    KERNELS=="pci0000:00"
    SUBSYSTEMS==""
    DRIVERS==""

RicardoEPRodrigues, 2020-06-19:
This is going to be a long text. Sorry in advance.

So, the `DRIVERS=="hid-generic"` changes to `DRIVERS=="magicmouse"` when the driver is loaded. My guess is that for some reason on ElementaryOS when the driver changes it no longer loads the generic mouse driver.

I think you can try this out the following way:

1. Turn off the mouse
1. Use **your `udev` rule with the `DRIVERS=="hid-generic"`** or **my `udev` rule with the `DRIVERS=="hid-generic|magicmouse"`**
1. `sudo ./install-fix.sh`
1. Reboot the system
1. Turn on the mouse and it should connect and work correctly
1. Turn the mouse off and on and the mouse connects
1. If you used your `udev` rule, the mouse should work without scrolling
1. If you used my `udev` rule, the mouse should not work except for scrolling

Sorry for the work, but can you check if this happens?

---

Another thing also came to mind from your previous comments. First off, it seems you only have one mouse connect and it is the magic mouse (because the mouse is assigned `mouse0`), correct?

Second, `lsmod | grep hid` seems similar when it is working, I want to see the output of this when the **scrolling works, but the mouse doesn't**. Can you get it for me?

Why am I asking these questions? I think that when the `magicmouse` driver is loaded it no longer loads the `hid-generic` if no more mice are connected, thus the mouse doesn't work but the scroll does. Therefore, if you have an extra mouse around can you give this a try? Plug in that mouse and then (with my `udev` rule) connect the magic mouse. Let's see what happens.

RicardoEPRodrigues, 2020-06-19:
Sorry for the spam.

I now believe this is not an ElementaryOS issue, but rather I believe this is an issue that happens only when you don't have another mouse connected. (Can you give me the output of `ls /dev/input`)

I made a commit to my repo. Before loading the magic mouse driver, the script now also loads the `hid-generic` driver. @ralphsc can you please give it a try?

🤞Let's hope it works!

ralphsc, 2020-06-19:
OK, I'll give it a shot and report back. Here's you go:

ls /dev/input/
by-id    event1   event12  event15  event3  event6  event9  mouse1
by-path  event10  event13  event16  event4  event7  mice
event0   event11  event14  event2   event5  event8  mouse0

ralphsc, 2020-06-19:
I only have one mouse and of course the trackpad on the laptop. I don't have another mouse at the moment. I tried your changes to the script, mine looks like this now:

#!/bin/sh

FILE=/tmp/magicmouse-driveload

reload() {
    if [ ! -f "$FILE" ]; then
        touch $FILE

        modprobe -r hid_magicmouse
        modprobe -a hid-generic
        sleep 2
        insmod /opt/magic-mouse-fix/hid-magicmouse.ko \
            scroll_acceleration=1 \
            scroll_speed=35 \
            emulate_3button=0 \
            middle_click_3finger=0

        sleep 2
        rm -f "$FILE"

    fi
}

reload &

The udev rule just will not load the driver with "magicmouse" in it. my current working rule is this:

SUBSYSTEM=="input", \
    KERNEL=="mouse*", \
    DRIVER=="", \
    SUBSYSTEMS=="hid", \
    KERNELS=="0005:004C:0269*", \
    DRIVERS=="hid-generic", \
    ACTION=="add", \
    SYMLINK+="input/magicmouse-%k", \
    RUN+="/opt/magic-mouse-fix/magic-mouse-2-add.sh"

Thanks for all your help with this.

RicardoEPRodrigues, 2020-06-19:
Use my `udev` rule. With the `DRIVERS=="hid-generic|magicmouse".

ralphsc, 2020-06-19:
I've tried it that way, it fails to load the driver.

RicardoEPRodrigues, 2020-06-19:
I believe your version of `udev` doesn't accept the `|`, but I can't be sure. I suggest creating 2 rules, one with `DRIVERS=="hid-generic"` and another with `DRIVERS=="magicmouse"`. The rest would be the same. Can you try it out?

Also, can you get to the situation where the mouse connects but does nothing and send me the `lsmod | grep hid`?

landon-hemsley, 2020-10-24:
I have fought this for a day. I've managed to get basically nowhere. I've tried every reasonable variation I can think of, but ultimately, this just isn't working for me. I tried removing the line where `hid-generic` is added to the kernel modules. I tried altering the udev rule like this: `hid-generic|magicmouse` --> `magicmouse`, and also `hid-generic`. I tried making two rules. I restarted my system so many times. I'm pretty much at my wits end.

I'm on ubuntu 18.04, kernel version 5.4.0-52-generic.

I have a lot of different outputs, but this one seems to be the one that is most common. Nothing at all works with the mouse, neither motion nor scrolling. I am using the udev rule and the shell script at the head of the `magicmouse2` branch of this repo. Here's the output when I connect the mouse, having run `install-fix.sh`.

```
[   412.262] (II) config/udev: Adding input device Landon Hemsley’s Mouse (/dev/input/event29)
[   412.262] (**) Landon Hemsley’s Mouse: Applying InputClass "libinput pointer catchall"
[   412.262] (II) Using input driver 'libinput' for 'Landon Hemsley’s Mouse'
[   412.287] (EE) systemd-logind: failed to take device /dev/input/event29: No such device
[   412.288] (**) Landon Hemsley’s Mouse: always reports core events
[   412.288] (**) Option "Device" "/dev/input/event29"
[   412.288] (**) Option "_source" "server/udev"
[   412.288] (EE) client bug: Invalid path /dev/input/event29
[   412.288] (EE) libinput: Landon Hemsley’s Mouse: Failed to create a device for /dev/input/event29
[   412.288] (EE) PreInit returned 2 for "Landon Hemsley’s Mouse"
[   412.288] (II) UnloadModule: "libinput"
[   416.795] (II) config/udev: Adding input device (unnamed) (/dev/input/mouse3)
[   416.796] (II) No input driver specified, ignoring this device.
[   416.796] (II) This device may have been added with another device file.
```
This is the `udevadm info` on this device:
```
  looking at device '/devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:256/0005:004C:0269.0005/input/input866/mouse3':
    KERNEL=="mouse3"
    SUBSYSTEM=="input"
    DRIVER==""

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:256/0005:004C:0269.0005/input/input866':
    KERNELS=="input866"
    SUBSYSTEMS=="input"
    DRIVERS==""
    ATTRS{phys}=="28:7f:cf:94:dc:01"
    ATTRS{properties}=="0"
    ATTRS{uniq}=="30:d9:d9:8a:27:47"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:256/0005:004C:0269.0005':
    KERNELS=="0005:004C:0269.0005"
    SUBSYSTEMS=="hid"
    DRIVERS=="hid-generic"
    ATTRS{country}=="00"
.
.
.
.
etc.
```

I have verified that in the `/dev/input` folder, a listing for `magicmouse-mouse3` exists, and its a symlink to `mouse3`.
Best I can tell is that the rule is triggered, but something in the shell script isn't playing nice for me. (`PreInit returned 2` is the biggest clue for me on that front)

I'm not using anything special in terms of system configuration. My system is pretty vanilla. I can't help but wonder if it's because I'm still on 18.04. I learned earlier today that ubuntu 20.04.1 LTS is available. I can't now, but at some point in the near future, I'll upgrade the OS and maybe try again. If there's anything y'all know of or can suggest or recommend, I would appreciate it.

landon-hemsley, 2020-10-24:
Sorry for spam. I also found this. Perhaps I should look at altering my xorg conf files to recogize this particular device?

https://unix.stackexchange.com/questions/422126/xorg-reports-no-input-driver-specified-ignoring-this-device-am-i-using-the

RicardoEPRodrigues, 2020-10-25:
@landon-hemsley you might be on to something. I know it works on newer systems, but I also have complaints about it not working on ElementaryOS 5 which is based on Ubuntu 18.04 if I'm not wrong.

After a quick search I found this sentence online: 

> Unless `xorg.conf` has manual configuration, X.org uses udev for detecting input devices, and udev properties for configuring them.

It come from [Kubuntu Wiki](https://wiki.kubuntu.org/X/InputConfiguration) and it is a worthy read.

I tested it on Pop_OS 20.04 and even with an X.org file it works fine, yet my guess is that it does not apply to older X.org versions.

---

In the same page, I also found this section interesting:

> ## Driver assignment
>
> Most input devices like simple mice, keyboards, lid switches, etc. are supported by the general "evdev" X.org input driver, so /lib/udev/rules.d/65-xorg-evdev.rules sets ENV{x11_driver}="evdev" by default.
>
> Later rules (i. e. with a prefix higher than 65) can change the property to use a different driver. 66-xorg-synaptics.rules is a prominent example for using the "synaptics" driver for touchpads and setting ENV{x11_driver}="synaptics" for devices which have ID_INPUT_TOUCHPAD.

Which got me thinking if changing the name of the `udev` file from `10-magicmouse.rules` to `90-magicmouse.rules` would make a difference in your systems.

_PS: I'm no expert and I'm learning as I go, so for me it is a lot of trial and error. I wish I knew more to be able to great an actual great driver._

landon-hemsley, 2020-11-26:
FWIW, and just to close the loop on this, I upgraded my system to 20.04.1, went through the process of re-installing the driver and applied @RicardoEPRodrigues 's fix. It worked! So, my recommendation: Upgrade! lol.

akozyreva, 2020-12-01:
trying on Manjaro - scrolling worked fine ~ 5 hours and after that doesn't work again. I followed all steps from instruction from forked README. @RicardoEPRodrigues any ideas, what can be wrong?

RicardoEPRodrigues, 2020-12-03:
@akozyreva hey! I don't know if I commented this or not, but normally after some OS updates the drivers stops working and I have to "reinstall" it.

Normally running `sudo ./install-fix.sh` and rebooting does the job.

KristijanZic, 2020-12-10:
It worked for me too but I find that often the right mouse button is registered as a left mouse button or some time both.
I noticed that when browsing the web in Firefox. If I right click a link to show the context menu and go "open in a new tab" and left click it, it opens it both in the current one and in a new tab. Often times when I right click the link it just follows it immediately as if I clicked the left button.

Also when selecting text and trying to right click it. It just deselects it and moves the text cursor just as a left click would. 

I'm on Ubuntu 20.04 is anybody able to reproduce this?