mwyborski/Linux-Magic-Trackpad-2-Driver #40: inconsistent USB vs bluetooth

esfourteen, 2019-05-11:
When using the trackpad via bluetooth, a quirks override is still necessary to fix the pressure issues. Additionally, for some reason palm rejection doesn't work at all (just picks up palm as mouse movements).

When using the trackpad via USB, pressures work as expected without any quirks and palm rejection works reliably.

Tested this with both live USB ubuntu 19.04 (kernel 5.0) and elementary juno (kernel 5.0.11), same issues.

esfourteen, 2019-05-11:
additionally, xinput sees the device as 'Apple Inc. Magic Trackpad 2' with usb or bluetooth, but shows different product ID's.

When connected via usb the following additional input properties are present:

`libinput Disable While Typing Enabled (315):    1
libinput Disable While Typing Enabled Default (316):    1`

these are not present when connected via bluetooth.

esfourteen, 2019-05-11:
USB:
[image]

Those quirks listed appear to be the default USB quirks defined for apple touchpads:

    [Apple Touchpads USB]
    MatchVendor=0x05AC
    MatchBus=usb
    MatchUdevType=touchpad
    ModelAppleTouchpad=1
    AttrSizeHint=104x75
    AttrTouchSizeRange=150:130
    AttrPalmSizeThreshold=800

Bluetooth:
[image]

note how theres no quirks listed, however copying the quirks shown above into local overrides only fixes the pressure which allows mouse movement and scrolling. palm detection still doesn't work

dos1, 2019-05-11:
Patches necessary for it to work out-of-the-box are already in both systemd (https://github.com/systemd/systemd/pull/12282, merged in v242) and libinput (https://gitlab.freedesktop.org/libinput/libinput/merge_requests/214, merged in 1.13.2). You have to wait until your distribution ships them or apply those fixes by yourself.

dos1, 2019-05-11:
BTW. Palm rejection gets automatically disabled for external devices, so the correct behavior is to have it disabled for both Bluetooth and USB (it wasn't disabled for USB, because it was incorrectly marked as internal instead).

esfourteen, 2019-05-11:
@dos1 i see, that makes sense. what would be the correct way to enable palm detection for bt (or usb when patch is merged)?

dos1, 2019-05-11:
Not sure if libinput allows you to reenable palm detection for external touchpads, I guess it doesn't, but you can always use a local udev-hwdb override to mark the touchpad as internal (see the systemd PR for reference).

esfourteen, 2019-05-11:
@dos1 ah yep, that works. thanks!

closed 2019-05-11 by esfourteen