mwyborski/Linux-Magic-Trackpad-2-Driver #42: Update README updates for easier to understand

wolftune, 2019-06-27:
its not clear which update suggestions apply to which scenarios or whether they are alternatives to one another or to be used in conjunction.

It would be good to update the README to make it clearer.

Also, the basic what-this-is should go above the updates sections.

robotrovsky, 2019-06-28:
@wolftune this is a open source project and lives from the community. If you like to reorganize/improve the README, then i am happy to merge your pull-request. 

The driver is part of all kernels starting from 4.20, so the suggested update is to run a recent kernel.

closed 2019-06-28 by robotrovsky

wolftune, 2019-06-28:
@robotrovsky I'm well-aware of open source development; I didn't open the issue to ask or demand that others do work. Often the way to clarify about issues is to discuss the concern and get consensus first rather than fly-by PRs (which presume your time in making sense of them).

I'd be happy to submit a PR. I'm not totally clear about how to describe the status though. Since the README "updates" sections happened *after* the 4.20 merge of the driver, are they also merged or are they potentially relevant regardless of getting newer driver?

curiousercreative, 2020-10-29:
Sad to see this issue closed, I don't even know how to install the thing and verify it's working!

mwyborski, 2020-10-29:
If you dont know how to verify that your trackpad is working maybe Linux isn’t for you ;-) Otherwise any distribution with a kernel newer than 4.20 should work out of the box. Please give more constructive crticism or even better a pull request with a better README.MD. What do you expect in the Readme that isn’t there?

@wolftune the only relevant update after the merge was how to make it usable on Xubuntu. 

Ubuntu works out of the box since 19.10. Usually people can ignore this repository as the driver is part of all recent distributions.

wolftune, 2020-10-30:
Yes indeed, this is not a repo for end-users basically. End users should just have things working and not have to do anything.

I still have issue #37 happening, but I just have this routine now of toggling on and off the touchpad whenever I wake the computer. No huge problem, though getting that fixed would be nice. But whenever that's fixed it'll get to the kernel and still there's no end-user anything that normal users should be doing in terms of installing or checking anything.