mwyborski/Linux-Magic-Trackpad-2-Driver #45: Windows Support

brave3d, 2019-08-03:
Hello everyone, I know it may not be a related issue, but I wonder if anyone was able or has information on how to make this as starting point to support Windows, IE. Make a reliable driver for apple magic mouse on Windows, the issue with the official driver is that it doesn't support scrolling, but only mouse movement and it disconnects frequently, I made tiny windows service that prevents disconnection and it's working fine but the scrolling issue remains.
I have tried Magic Mouse Utilities also, which solves this problem, but it's closed source and paid program, you have to pay for a license per device per year, or you may buy a multiyear license which is a joke.
So if anyone has a hint or a starting point we may collaborate to solve this issue. Thank you.

Gintoki1ku, 2020-01-15:
> Hello everyone, I know it may not be a related issue, but I wonder if anyone was able or has information on how to make this as starting point to support Windows, IE. Make a reliable driver for apple magic mouse on Windows, the issue with the official driver is that it doesn't support scrolling, but only mouse movement and it disconnects frequently, I made a tiny windows service that prevents disconnection and it's working fine but the scrolling issue remains.
> I have tried Magic Mouse Utilities also, which solves this problem, but it's closed source and paid program, you have to pay for a license per device per year, or you may buy a multiyear license which is a joke.
> So if anyone has a hint or a starting point we may collaborate to solve this issue. Thank you.

You can check out this https://github.com/imbushuo/mac-precision-touchpad. I just installed it on Windows and enabled the touchpad in settings which works perfect.

brave3d, 2020-01-15:
> You can check out this https://github.com/imbushuo/mac-precision-touchpad. I just installed it on Windows and enabled the touchpad in settings which works perfect.

Seems like this is designed for touchpad, have you tried it with magic mouse?

Ansbot, 2020-01-15:
> Seems like this is designed for touchpad, have you tried it with magic mouse?

My apologies. I thought you need support for trackpad and I don't have magic mouse😂. Maybe you can try the apple bootcamp driver.

Gintoki1ku, 2020-01-15:
> Seems like this is designed for touchpad, have you tried it with magic mouse?

My apologies. I thought you want support for trackpad and I don't have magic mouse.