mwyborski/Linux-Magic-Trackpad-2-Driver #47: Support for Secondary Click

hiradyazdan, 2019-10-15:
Hi, I was wondering if this driver can support a secondary click that can be activated and have option to choose to click in bottom right corner as well? If not, it would be great to support it.

Thanks

dos1, 2019-10-15:
Unless you mean something like "force touch", it's not the driver's job. You should rather look at something like libinput instead. Secondary button clicks work just fine for me with two fingers, and having virtual buttons in bottom corners worked fine after applying appropiate libinput quirks last time I checked.

hiradyazdan, 2019-10-15:
Oh! Thanks, I didn't realize that. I guess now I have to figure out how to use the `libinput`'s button areas and clickfinger behavior, right?

hiradyazdan, 2019-10-23:
FYI, `libinput` quirks on their own does not take permanent effect and solves this problem. It is the desktop environment which overrides the settings. So in my case was gnome and following command worked permanently:

```shell
gsettings set org.gnome.desktop.peripherals.touchpad click-method 'areas'
```

closed 2019-10-23 by hiradyazdan