mwyborski/Linux-Magic-Trackpad-2-Driver #50: Drag stops when pressing a bit harder while dragging

bitfabrikken, 2019-12-02:
Does anyone know how to make the Magic Trackpad 2 keep dragging, even though you're pressing hard? As it is now, I have to drag with a feather-light touch, and it's hard to get used to.

I've tried changing AttrPressureRange, but it doesn't seem to do anything.
Perhaps someone knows what to change in regards to dragging pressure?

When I press hard while dragging, the pointer just stops moving completely.

Debian Buster, 5.3.0-0.bpo.2-amd64
xorg-server 2:1.20.4-1

xorg log when plugging in:
```
[  5076.799] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/mouse0)
[  5076.799] (II) No input driver specified, ignoring this device.
[  5076.799] (II) This device may have been added with another device file.
[  5076.989] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event1)
[  5076.989] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput touchpad catchall"
[  5076.989] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "Touchpads"
[  5076.989] (II) Using input driver 'libinput' for 'Apple Inc. Magic Trackpad 2'
[  5076.989] (**) Apple Inc. Magic Trackpad 2: always reports core events
[  5076.989] (**) Option "Device" "/dev/input/event1"
[  5076.989] (**) Option "_source" "server/udev"
[  5077.049] (II) event1  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Touchpad
[  5077.050] (II) event1  - Apple Inc. Magic Trackpad 2: device is a touchpad
[  5077.050] (II) event1  - Apple Inc. Magic Trackpad 2: device removed
[  5077.080] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:01.1/0000:01:00.0/usb1/1-7/1-7:1.1/0003:05AC:0265.0010/input/input31/event1"
[  5077.080] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: TOUCHPAD, id 8)
[  5077.082] (**) Option "AccelerationScheme" "none"
[  5077.084] (**) Apple Inc. Magic Trackpad 2: (accel) selected scheme none/0
[  5077.084] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[  5077.084] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[  5077.145] (II) event1  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Touchpad
[  5077.145] (II) event1  - Apple Inc. Magic Trackpad 2: device is a touchpad
```

xinput list-props 8
```
Device 'Apple Inc. Magic Trackpad 2':
        Device Enabled (155):   1
        Coordinate Transformation Matrix (157): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Tapping Enabled (291): 0
        libinput Tapping Enabled Default (292): 0
        libinput Tapping Drag Enabled (293):    1
        libinput Tapping Drag Enabled Default (294):    1
        libinput Tapping Drag Lock Enabled (295):       1
        libinput Tapping Drag Lock Enabled Default (296):       0
        libinput Tapping Button Mapping Enabled (297):  0, 1
        libinput Tapping Button Mapping Default (298):  1, 0
        libinput Natural Scrolling Enabled (299):       1
        libinput Natural Scrolling Enabled Default (300):       0
        libinput Disable While Typing Enabled (301):    1
        libinput Disable While Typing Enabled Default (302):    1
        libinput Scroll Methods Available (303):        1, 1, 0
        libinput Scroll Method Enabled (304):   1, 0, 0
        libinput Scroll Method Enabled Default (305):   1, 0, 0
        libinput Click Methods Available (306): 1, 1
        libinput Click Method Enabled (307):    0, 1
        libinput Click Method Enabled Default (308):    0, 1
        libinput Middle Emulation Enabled (309):        0
        libinput Middle Emulation Enabled Default (310):        0
        libinput Accel Speed (311):     0.155378
        libinput Accel Speed Default (312):     0.000000
        libinput Left Handed Enabled (313):     0
        libinput Left Handed Enabled Default (314):     0
        libinput Send Events Modes Available (276):     1, 1
        libinput Send Events Mode Enabled (277):        0, 0
        libinput Send Events Mode Enabled Default (278):        0, 0
        Device Node (279):      "/dev/input/event1"
        Device Product ID (280):        1452, 613
        libinput Drag Lock Buttons (315):       <no items>
        libinput Horizontal Scroll Enabled (316):       1
```

on x start I run this:
```
xinput set-prop 8 "libinput Tapping Enabled" 1
xinput set-prop 8 "libinput Disable While Typing Enabled" 0 
xinput set-prop 8 "libinput Accel Speed" 0.2
xinput set-prop 8 "libinput Middle Emulation Enabled" 1
```

robotrovsky, 2019-12-02:
Which desktop are you using? For XFCE create the file like explained in the README.MD

bitfabrikken, 2019-12-02:
@robotrovsky KDE Plasma

robotrovsky, 2019-12-02:
I haven't used KDE Plasma with the MT2 yet. But i think it might work in Kubuntu 19.10 as it is working in the standard Ubuntu as well. You could check in a virtual machine if it works for you in Kubuntu and compare the configuration with your own.

bitfabrikken, 2019-12-04:
@robotrovsky I've tried in Kubuntu 19.10, and yep, it works correctly there.
Only differences are:
Kubuntu libinput 1.14.1 vs 1.12.6-2 on debian
Kubuntu kernel 5.3.0-24-generic vs 5.3.9-2bpo10+1 (2019-11-13)
Not too sure what to try next..

robotrovsky, 2019-12-04:
@bitfabrikken great, so we are a step closer ;-)
The kernel version should be no problem as long as magicmouse driver (contains MT2) is included. About the libinput i am not sure, but as your version is higher i think it neither is the problem.
You could now check the udev rule and the X / KDE configuration. I guess that is where the difference to your system is. I think the libinput driver is loaded but not configured correct. 

I suppose you are using X11 with KDE. Have you tried creating the following file:
` /usr/share/X11/xorg.conf.d/90-magictrackpad.conf`

```
Section "InputClass"
  Identifier      "Touchpads"
  Driver          "libinput"
  MatchProduct    "Apple Inc. Magic Trackpad 2"
  MatchDevicePath "/dev/input/event*"
EndSection
```

bitfabrikken, 2019-12-04:
@robotrovsky closing in I hope :) I'll give some more verbose info:

I was on debian's stable kernel for a while (4.19.67-2+deb10u2), where I tried using this repos (driver (via make, rmmod, insmod), but it didn't work and so switched to the backported kernel from testing (5.3.9-2~bpo10+1).

Now it works, but with the mentioned error where it stops moving the cursor when pressing too hard.

The 90-magictrackpad.conf file is also present (am running X11, X.Org version: 1.20.4)
I've also tried with the /etc/libinput/local-overrides.quirks file and many, many different values for stuff. Also logging out of X for every change, doesn't seem to apply it correctly otherwise.

When comparing the Xorg logs for both systems, when MT2 is plugged in, there's 6 lines with "Applying InputClass" on the debian, that do not happen on the Kubuntu. 
Do you think the problem could be there?

I've noticed the error doesn't happen if I try the DKMS thing from the README.md, although it does seem to freeze the computer sometimes.

Plugging in via USB (haven't tried Bluetooth as I want USB to work first) on the real debian machine, Xorg.log:
```
[ 15940.014] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/mouse0)
[ 15940.014] (**) Apple Inc. Magic Trackpad 2: Ignoring device from InputClass "touchpad ignore duplicates"
[ 15940.221] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event1)
**[ 15940.221] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput touchpad catchall"
[ 15940.221] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "Multitouch Touchpad"
[ 15940.221] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "touchpad catchall"
[ 15940.221] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "Default clickpad buttons"
[ 15940.221] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "Disable clickpad buttons on Apple touchpads"**
[ 15940.221] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "Touchpads"
[ 15940.221] (II) Using input driver 'libinput' for 'Apple Inc. Magic Trackpad 2'
[ 15940.221] (**) Apple Inc. Magic Trackpad 2: always reports core events
[ 15940.221] (**) Option "Device" "/dev/input/event1"
[ 15940.221] (**) Option "_source" "server/udev"
[ 15940.280] (II) event1  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Touchpad
[ 15940.280] (II) event1  - Apple Inc. Magic Trackpad 2: device is a touchpad
[ 15940.280] (II) event1  - Apple Inc. Magic Trackpad 2: device removed
[ 15940.312] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:01.1/0000:01:00.0/usb1/1-7/1-7:1.1/0003:05AC:0265.001D/input/input39/event1"
[ 15940.312] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: TOUCHPAD, id 8)
[ 15940.313] (**) Option "AccelerationScheme" "none"
[ 15940.314] (**) Apple Inc. Magic Trackpad 2: (accel) selected scheme none/0
[ 15940.314] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[ 15940.314] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[ 15940.372] (II) event1  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Touchpad
[ 15940.372] (II) event1  - Apple Inc. Magic Trackpad 2: device is a touchpad

```

Plugging in via usb in the virtual machine Kubuntu, Xorg log:
```
[    78.473] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/mouse0)
[    78.473] (II) No input driver specified, ignoring this device.
[    78.473] (II) This device may have been added with another device file.
[    78.637] (II) config/udev: Adding input device Apple Inc. Magic Trackpad 2 (/dev/input/event2)
[    78.637] (**) Apple Inc. Magic Trackpad 2: Applying InputClass "libinput touchpad catchall"
[    78.637] (II) Using input driver 'libinput' for 'Apple Inc. Magic Trackpad 2'
[    78.637] (**) Apple Inc. Magic Trackpad 2: always reports core events
[    78.637] (**) Option "Device" "/dev/input/event2"
[    78.638] (**) Option "_source" "server/udev"
[    78.729] (II) event2  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Touchpad
[    78.730] (II) event2  - Apple Inc. Magic Trackpad 2: device is a touchpad
[    78.730] (II) event2  - Apple Inc. Magic Trackpad 2: device removed
[    78.769] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:05.0/usb2/2-1/2-1:1.1/0003:05AC:0265.0006/input/input8/event2"
[    78.769] (II) XINPUT: Adding extended input device "Apple Inc. Magic Trackpad 2" (type: TOUCHPAD, id 7)
[    78.771] (**) Option "AccelerationScheme" "none"
[    78.772] (**) Apple Inc. Magic Trackpad 2: (accel) selected scheme none/0
[    78.772] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration factor: 2.000
[    78.772] (**) Apple Inc. Magic Trackpad 2: (accel) acceleration threshold: 4
[    78.833] (II) event2  - Apple Inc. Magic Trackpad 2: is tagged by udev as: Touchpad
[    78.834] (II) event2  - Apple Inc. Magic Trackpad 2: device is a touchpad

```

robotrovsky, 2019-12-04:
@bitfabrikken i think this is a X11 misconfiguration. I would compare the xorg configuration files. I am sorry that i can't help you more. But i agree that you should get rid of the "Applying InputClass" lines in dmesg. Maybe when another driver gets loaded the values/thresholds get changed or something.

bitfabrikken, 2019-12-04:
Thanks @robotrovsky, I'll have a look in sec.

I just noticed something else that perhaps could pinpoint?
When using "libinput measure touchpad-pressure", I see that "palm" is active when I reach ~130 pressure. This appears to be exactly when the mouse cursor stops moving. The hardware "click" by the motor appears around 90.

I've currently got this in quirks:
ModelAppleTouchpad=1
AttrSizeHint=104x75
AttrTouchSizeRange=150:130
AttrPalmSizeThreshold=800

bitfabrikken, 2019-12-05:
@robotrovsky I ended up going to debian testing instead of stable. Works out of the box now! Haven't run debian for desktop in 20ish years, and apparantly stable is waaaay behind. Anyways, appreciate all your help the last few days!

Unrelated, but perhaps you can point me in the right direction to have three-finger touch move windows? Am also wanting to make four fingers resize windows. You probably know if any of those things is even doable :)

closed 2019-12-05 by bitfabrikken

robotrovsky, 2019-12-05:
@bitfabrikken you are welcome. Regarding three finger drag look into mtrack driver instead of using libinput. 
https://github.com/BlueDragonX/xf86-input-mtrack

bitfabrikken, 2019-12-05:
@robotrovsky thanks, but I've tried mtrack and it's too laggy when scrolling for me :)

robotrovsky, 2019-12-05:
@bitfabrikken i agree. Three finger-drag works very well, but the scrolling is ugly..