mwyborski/Linux-Magic-Trackpad-2-Driver #51: Driver not loading on startup

AlanBStone, 2020-01-06:
I've installed the driver according to the instructions.  Everything works great, however when I restart the system for any reason, the driver is not loaded at startup.  I am forced to load it manually, then all is great again.  I'm not exactly sure what is going on.  Thanks.

danielkappelle, 2020-03-23:
I'm using the fork from rohitpid (https://github.com/rohitpid/Linux-Magic-Trackpad-2-Driver) to support Magic Mouse 2, but I have a similar issue after reboot or reconnecting the mouse. The module gets loaded, but I have to reload the module in order for scrolling to work. I can't figure out what the difference is though...

The module is loaded (after connecting the mouse), but scrolling does not work...

```
[root@kappellearch hid]# lsmod | grep hid_magicmouse
hid_magicmouse         20480  0
hid                   147456  5 i2c_hid,hidp,hid_multitouch,hid_generic,hid_magicmouse
[root@kappellearch hid]# modinfo hid_magicmouse
filename:       /lib/modules/5.5.9-arch1-2/kernel/drivers/hid/hid-magicmouse.ko.xz
license:        GPL
srcversion:     3D17F9A341D5471ADBEC2A6
alias:          hid:b0003g*v000005ACp00000265
alias:          hid:b0005g*v0000004Cp00000265
alias:          hid:b0005g*v000005ACp0000030E
alias:          hid:b0005g*v0000004Cp00000269
alias:          hid:b0005g*v000005ACp0000030D
depends:        hid
retpoline:      Y
name:           hid_magicmouse
vermagic:       5.5.9-arch1-2 SMP preempt mod_unload 
parm:           emulate_3button:Emulate a middle button (bool)
parm:           middle_click_3finger:Use 3 finger click to emulate middle button (bool)
parm:           emulate_scroll_wheel:Emulate a scroll wheel (bool)
parm:           scroll_speed:Scroll speed, value from 0 (slow) to 63 (fast)
parm:           scroll_acceleration:Accelerate sequential scroll events (bool)
parm:           report_undeciphered:Report undeciphered multi-touch state field using a MSC_RAW event (bool)
```

If I reload the module, it does work, i.e. if I
```
# rmmod hid_magicmouse
# modprobe hid_magicmouse
```

If anyone has any ideas what's causing this or how I can figure out what is different when the module is loaded automatically, please let me know :)

eaudet, 2020-04-01:
After a restart, I also need to do this
`# rmmod hid_magicmouse
# modprobe hid_magicmouse`

nmosto, 2020-06-29:
I'm having the same issue with scrolling not working with the mouse.
Another question I have is concerns the difference between insmod and modprobe (I see modprobe uses insmod). After removing hid_magicmouse
`sudo rmmod hid_magicmouse`
why wont it work if I load it back with insmod?
'sudo insmod /lib/modules/5.3.0-61-generic/kernel/drivers/hid/hid-magicmouse.ko'

I see that it is loaded
'$ lsmod | grep hid_magicmouse
hid_magicmouse         20480  0
hid                   126976  7 hidp,usbhid,hid_apple,hid_generic,hid_logitech_dj,hid_logitech_hidpp,hid_magicmouse
'

kmani314, 2020-12-05:
I'm also having this issue with the magicmouse2 fork on archlinux. Everything seems to be fine when I reload the driver, but I can't get it to start on boot by putting it in `/etc/modules.load.d`.