mwyborski/Linux-Magic-Trackpad-2-Driver #53: not working with Ubuntu 20.04

christianlaurent1992, 2020-04-08:
Hello everyone, 

This code has always just worked for me with magic mouse 2 and is excelent- I have reinstalled after upgrading to ubuntu 20.04 and performed the troubleshooting steps on the read.me, to no avail:

```
[   826.825] (**) rabbit’s Mouse: Applying InputClass "libinput pointer catchall"
[   826.825] (II) Using input driver 'libinput' for 'rabbit’s Mouse'
```

`sudo insmod ./hid-magicmouse.ko`  executes without an error

I've never dabbled in any driver development before and am probably too newbie to solve this myself but i'm happy to be the guinea pig before anybody elses system breaks...

frankgould, 2020-05-18:
Hi christianlaurent1992, I was able to install the magic mouse (2 I believe) but it took a few different approaches until it finally worked for me. I haven't gone back to confirm all is needed but I can list what I tried that is now operational with horizontal/vertical scroll. Maybe one of these will work for you. The two links below were most helpful.

https://www.raspberrypi.org/forums/viewtopic.php?t=272419
http://sneclacson.blogspot.com/2016/09/using-apple-magic-mouse-with-ubuntu-1604.html

Install/configure as follows:
```
sudo apt install pi-bluetooth
sudo apt install dkms
git clone https://github.com/rohitpid/Linux-Magic-Trackpad-2-Driver.git
cd Linux-Magic-Trackpad-2-Driver/scripts
chmod u+x install.sh
sudo ./install.sh 
```
Edit /boot/firmware/syscfg.txt
`sudo nano /boot/firmware/syscfg.txt`

Add the following line to the end of the file:
`include btcfg.txt`

Execute the following commands:
```
sudo rmmod hid_magicmouse
sudo modprobe hid_magicmouse emulate_3button=0 scroll_acceleration=1 scroll_speed=55
```
# vvvv The below may not be necessary vvvvv
Edit /usr/share/X11/xorg.conf.d/90-magictrackpad.conf file:
`sudo nano /usr/share/X11/xorg.conf.d/90-magictrackpad.conf
`
Add the following lines to the new file:
```
Section "InputClass"
  Identifier      "Touchpads"
  Driver          "libinput"
  MatchProduct    "Apple Inc. Magic Trackpad 2"
  MatchDevicePath "/dev/input/event*"
EndSection
```
# ^^^^ The above may not be necessary ^^^^^
To get the results of the installation and further configure mouse:
Get `systool` command by installing the following:
`sudo apt install sysfsutils`

```
$ systool -avm hid_magicmouse
Module = "hid_magicmouse"

  Attributes:
    coresize            = "20480"
    initsize            = "0"
    initstate           = "live"
    refcnt              = "0"
    srcversion          = "3D17F9A341D5471ADBEC2A6"
    taint               = "OE"
    uevent              = <store method only>

  Parameters:
    emulate_3button     = "N"
    emulate_scroll_wheel= "Y"
    middle_click_3finger= "N"
    report_undeciphered = "N"
    scroll_acceleration = "Y"
    scroll_speed        = "55"

$ xinput list-props 11
Device 'Apple Magic Mouse':
	Device Enabled (116):	1
	Coordinate Transformation Matrix (117):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	libinput Natural Scrolling Enabled (246):	0
	libinput Natural Scrolling Enabled Default (247):	0
	libinput Scroll Methods Available (248):	0, 0, 1
	libinput Scroll Method Enabled (249):	0, 0, 0
	libinput Scroll Method Enabled Default (250):	0, 0, 0
	libinput Button Scrolling Button (251):	2
	libinput Button Scrolling Button Default (252):	2
	libinput Middle Emulation Enabled (253):	0
	libinput Middle Emulation Enabled Default (254):	0
	libinput Accel Speed (255):	-0.620000
	libinput Accel Speed Default (256):	0.000000
	libinput Accel Profiles Available (257):	1, 1
	libinput Accel Profile Enabled (258):	1, 0
	libinput Accel Profile Enabled Default (259):	1, 0
	libinput Calibration Matrix (464):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	libinput Calibration Matrix Default (465):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	libinput Left Handed Enabled (260):	0
	libinput Left Handed Enabled Default (261):	0
	libinput Send Events Modes Available (262):	1, 0
	libinput Send Events Mode Enabled (263):	0, 0
	libinput Send Events Mode Enabled Default (264):	0, 0
	Device Node (265):	"/dev/input/event3"
	Device Product ID (266):	76, 617
	libinput Drag Lock Buttons (267):	<no items>
	libinput Horizontal Scroll Enabled (268):	1
```

christianlaurent1992, 2020-05-18:
Thanks, I got it working too with some fiddling. But not without doing it manually after every reboot and tbh I haven’t noticed if horizontal works (but I think it does)

I have never gotten the 3 finger stuff to work, despite multiple attempts, have you?

https://www.raspberrypi.org/forums/viewtopic.php?t=272419
http://sneclacson.blogspot.com/2016/09/using-apple-magic-mouse-with-ubuntu-1604.html

frankgould, 2020-05-20:
Yes, same here with manual configuration execution after every reboot. There should be some way to have it execute during the boot sequence but in all my attempts, the script can't connect to the X server or display. Any suggestions there are welcome. Also part of my configuration script, I parse the xinput string to determine the mouse and command id to set the desired parameters.

As for three finger control, we don't know how that's supposed to work or why. So, I don't have anything to report about that feature. Horizontal scroll does work, however.

christianlaurent1992, 2020-05-20:
I had some success making a small bash script with an alias ‘scrolling my that went something like:

Cd directory
Make clean
Make
Rmmod hid Magic Mouse
Insmod hid Magic Mouse
Cowsay keep scrolling Mooooooooo

But that broke so I’ve taken to doing it manually for a while...

I guess if you got that working the you could execute the same script on start up?

Aurrix, 2020-08-02:
Hi Everyone,

I followed the instillation guidelines but it scrolling still doesn't work. I noticed that kernel module is not loaded.
So fixes proposed by @frankgould didn't really work for me.