mwyborski/Linux-Magic-Trackpad-2-Driver #54: Abnormal behaviour after installing in Ubuntu 16.04

DeeperMind, 2020-08-21:
Following #16 , I reverted ecfde9f and installed the driver in Ubuntu 16.04. However, after installation, the magic trackpad does not work properly. I can only move the cursor when keeping pressing. And the scrolling does not work at all. Does anyone have any ideas?

mwyborski, 2020-08-21:
I would recommend updating Ubuntu. Since 19.04 it works out of the box.

allantokuda, 2021-01-02:
I see the same problem on Ubuntu 20.04 LTS.
Pressing, I can move the cursor. Otherwise no cursor movement.
Tried creating the two files listed in the README, `/usr/share/X11/xorg.conf.d/90-magictrackpad.conf` and `/etc/libinput/local-overrides.quirks` and restarting the trackpad device. No change.
Here is what I get from `libinput list-devices`:
```
Device:           Apple Inc. Magic Trackpad 2
Kernel:           /dev/input/event2
Group:            7
Seat:             seat0, default
Size:             162x115mm
Capabilities:     pointer gesture
Tap-to-click:     disabled
Tap-and-drag:     enabled
Tap drag lock:    disabled
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   *two-finger edge 
Click methods:    button-areas *clickfinger 
Disable-w-typing: n/a
Accel profiles:   none
Rotation:         n/a
```

allantokuda, 2021-01-02:
Fixed by restarting the system. Now able to move cursor and two finger scroll! :tada: I imagine the Xorg server needed the restart for the configuration to apply.