mwyborski/Linux-Magic-Trackpad-2-Driver #57: How to integrate usr/share/X11/xorg.conf.d/90-magictrackpad.conf

EarlOfBurton, 2020-11-02:
Hello,

I'm fairly new to the linux world. Thanks a lot for the repo! my trackpad2 works, but I don't know how to integrate the 90-magictrackpad.conf into my system. I copied it with sudo in the corresponding folder usr/share/X11/xorg.conf.d/90-magictrackpad.conf
And did a system reboot. But the tap to click function is not working.

Do I need to add the 90-magictrackpad.conf somewhere to be loaded?
Thanks for your help in advance.

adam-h, 2020-11-02:
You likely need to copy it to `/etc/X11/xorg.conf.d/90-magictrackpad.conf` as `/usr/share` is just where the template lives, `/etc` is where it will be loaded from.

EarlOfBurton, 2020-11-03:
Hi @adam-h , thanks for your reply and excuse my stupidity: in my /etc/X11/ there are just those folders:
```
app-defaults             rgb.txt  Xreset.d    Xsession.options
cursors                  xinit    Xresources  xsm
default-display-manager  xkb      Xsession    XvMCConfig
fonts                    Xreset   Xsession.d  Xwrapper.config
```

EarlOfBurton, 2020-11-03:
@adam-h sorry for disturbing you, I noticed that my ubuntu supports those features natively, no need for that script. thanks anyway.

closed 2020-11-03 by EarlOfBurton

PaulPCIO, 2020-12-23:
Is it normal that the device does not persist settings?

I had to edit my  [~/.xsessionrc](https://www.topbug.net/blog/2017/02/23/enable-natural-scrolling-for-trackpads-using-libinput/) to include the set prop, that way it will get set each time I restart.
```
xinput --set-prop "Illia Danko’s Mouse" "libinput Scroll Method Enabled" 0 0 1
```

But then I noticed that it does not persist if I reconnect the device, so I'm going to try to do something similar in the conf?