mwyborski/Linux-Magic-Trackpad-2-Driver #58: No input device for USB connection

jonjonw, 2020-11-10:
My Magic Trackpad 2 has been working off and on over bluetooth but it doesn't automatically connect on boot and occasionally disconnects and won't reconnect until I reboot and turn the device off and on.

For now I would prefer to avoid the issue by connecting over USB instead but it seems that no input event device is created for it over USB.

I can see it in lsusb.py (other devices filtered for easier reading):
```
% lsusb.py 
usb1              1d6b:0002 09 1IF  [USB 2.00,   480 Mbps,   0mA] (xhci-hcd 0000:00:14.0) hub
  ...
  1-10              05e3:0608 09 1IF  [USB 2.00,   480 Mbps, 100mA] (Genesys Logic, Inc. Hub) hub
    ...
    1-10.4            0a05:0002 09 1IF  [USB 1.10,    12 Mbps, 100mA] (Unknown Manufacturer) hub
      1-10.4.1          05f3:0081 09 1IF  [USB 1.10,    12 Mbps,  50mA] (PI Engineering Kinesis Keyboard Hub) hub
        ...
        1-10.4.1.4         05ac:0265 00 0IFs [USB 2.00,    12 Mbps,      ] (Apple Inc. Magic Trackpad 2 CC2033202L6J2XQA5) 
...
```

lsusb sees it as a DAC which seems a bit odd (perhaps because it can do haptic feedback?):
```
% lsusb | grep 05ac:0265
Bus 001 Device 021: ID 05ac:0265 Apple, Inc. USB Audio DAC
```

I've tried running evtest on every device in /dev/input/event* and touched the touchpad with no output. It seems as though no event driver is created for it over USB. Please let me know if you have any ideas or insights.

```
% ls -l /dev/input/by-path   
lrwxrwxrwx 1 root root  9 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.3.2:1.0-event-mouse -> ../event5
lrwxrwxrwx 1 root root  9 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.3.2:1.0-mouse -> ../mouse0
lrwxrwxrwx 1 root root  9 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.3.2:1.1-event -> ../event6
lrwxrwxrwx 1 root root 10 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.3.3:1.0-event-kbd -> ../event10
lrwxrwxrwx 1 root root 10 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.3.3:1.1-event -> ../event12
lrwxrwxrwx 1 root root 10 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.3.3:1.1-event-mouse -> ../event11
lrwxrwxrwx 1 root root  9 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.3.3:1.1-mouse -> ../mouse1
lrwxrwxrwx 1 root root 10 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.3.3:1.2-event -> ../event14
lrwxrwxrwx 1 root root 10 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.3.3:1.2-event-kbd -> ../event13
lrwxrwxrwx 1 root root  9 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.4.1.2:1.0-event-kbd -> ../event7
lrwxrwxrwx 1 root root  9 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.4.1.2:1.1-event -> ../event8
lrwxrwxrwx 1 root root 10 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.4.1.3:1.0-event-mouse -> ../event15
lrwxrwxrwx 1 root root  9 Nov  8 06:20 pci-0000:00:14.0-usb-0:10.4.1.3:1.0-mouse -> ../mouse2
lrwxrwxrwx 1 root root  9 Nov  8 06:20 pci-0000:00:14.0-usb-0:4:1.2-event -> ../event4
lrwxrwxrwx 1 root root 10 Nov  8 06:20 platform-pcspkr-event-spkr -> ../event16

% ls -l /dev/input/by-id  
lrwxrwxrwx 1 root root  9 Nov  8 06:20 usb-05f3_0007-event-if01 -> ../event8
lrwxrwxrwx 1 root root  9 Nov  8 06:20 usb-05f3_0007-event-kbd -> ../event7
lrwxrwxrwx 1 root root  9 Nov  8 06:20 usb-Burr-Brown_from_TI_USB_Audio_DAC-event-if02 -> ../event4
lrwxrwxrwx 1 root root 10 Nov  8 06:20 usb-Kingsis_Peripherals_Evoluent_VerticalMouse_3-event-mouse -> ../event15
lrwxrwxrwx 1 root root  9 Nov  8 06:20 usb-Kingsis_Peripherals_Evoluent_VerticalMouse_3-mouse -> ../mouse2
lrwxrwxrwx 1 root root  9 Nov  8 06:20 usb-Logitech_USB_Receiver-event-if01 -> ../event6
lrwxrwxrwx 1 root root  9 Nov  8 06:20 usb-Logitech_USB_Receiver-event-mouse -> ../event5
lrwxrwxrwx 1 root root  9 Nov  8 06:20 usb-Logitech_USB_Receiver-mouse -> ../mouse0
lrwxrwxrwx 1 root root 10 Nov  8 06:20 usb-Microsoft_Microsoft®_2.4GHz_Transceiver_v9.0-event-if01 -> ../event12
lrwxrwxrwx 1 root root 10 Nov  8 06:20 usb-Microsoft_Microsoft®_2.4GHz_Transceiver_v9.0-event-if02 -> ../event14
lrwxrwxrwx 1 root root 10 Nov  8 06:20 usb-Microsoft_Microsoft®_2.4GHz_Transceiver_v9.0-event-kbd -> ../event10
lrwxrwxrwx 1 root root 10 Nov  8 06:20 usb-Microsoft_Microsoft®_2.4GHz_Transceiver_v9.0-if01-event-mouse -> ../event11
lrwxrwxrwx 1 root root  9 Nov  8 06:20 usb-Microsoft_Microsoft®_2.4GHz_Transceiver_v9.0-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root 10 Nov  8 06:20 usb-Microsoft_Microsoft®_2.4GHz_Transceiver_v9.0-if02-event-kbd -> ../event13

% libinput --version   
1.16.3

% systemctl --version
systemd 246 (246.6-1-arch)
+PAM +AUDIT -SELINUX -IMA -APPARMOR +SMACK -SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +ZSTD +SECCOMP +BLKID +ELFUTILS +KMOD +IDN2 -IDN +PCRE2 default-hierarchy=hybrid

% uname -a             
Linux * 5.9.6-zen1-1-zen #1 ZEN SMP PREEMPT Thu, 05 Nov 2020 21:00:51 +0000 x86_64 GNU/Linux
```

mwyborski, 2020-12-04:
@jonjonw i have no idea what could be the reason. But maybe you can try Ubuntu in a virtual machine and cross compare?

jonjonw, 2021-01-06:
I didn't get a chance to set-up a Ubuntu VM but upgrading to linux kernel 5.10.3 seems to have revolved the problem for me.

closed 2021-01-06 by jonjonw