mwyborski/Linux-Magic-Trackpad-2-Driver #59: Don't scroll when moving the mouse

mimi89999, 2021-02-03:
Hello,

When moving the mouse, my fingers and hand are moving slightly on the mouse causing it to register scroll events that makes the page and the button I'm moving the cursor to move. This issue mostly occurs with very high scroll speeds like `scroll_speed=55`. With low scroll speeds, this does not happen, but the scroll is to slow for me.

Would it be possible to add a setting to stop sending scroll events when the mouse is moving?

Depau, 2021-02-08:
(unauthoritative answer)

I think you should ask the libinput team about this: https://gitlab.freedesktop.org/libinput/libinput

The driver's job is to detect input events from the hardware, it's libinput's job to interpret them and transform them into UI interactions.

mimi89999, 2021-02-08:
Hello,
I'm not sure if they would like to implement that since it's basically specific to the magic mouse. I don't know any other mouse where it would be possible to unintentionally scroll when moving it.

Depau, 2021-02-08:
libinput has device quirks for quirky devices.

vchernin, 2021-04-19:
@mimi89999 this other repo linked below seems to be specifically for the magic mouse 2 so you may have more luck there:

[https://github.com/RicardoEPRodrigues/magicmouse-hid](https://github.com/RicardoEPRodrigues/magicmouse-hid) 

Someone even sent some of these linked patches to the kernel mailing lists:
https://lkml.org/lkml/2021/3/27/209